jvde-github/AIS-catcher #1: Select SDR by serial number

skymaker-c2is, 2021-07-07:
Hi,

we just found your project and the message rate of our receiver increased by 100 % (have a look at the map in Kiel: https://adsb.chaos-consulting.de/ ) - awesome work!
One thing we noticed is, that we can't select the RTL-SDR by the serial number set with rtl_eeprom (e.g. 1000002). That makes setups with multiple SDRs not reboot-safe, since the device index can change after a reboot. Is it possible that you would implement this?

Thanks again for your work!
Lukas

jvde-github, 2021-07-07:
Hi Lukas, many thanks for sharing your first experience and results. I find it very encouraging so much appreciated!

I did not think about this before but your suggestion makes perfect sense to me. I now have uploaded a small fix that sorts the device list on serial number (AIS-catcher -l). This would make selecting the device with the d-switch a bit more deterministic. 

I will implement something soon where the serial number can be selected  on the command line and come back. That would be the elegant solution.

Thanks for your feedback.

Jasper

skymaker-c2is, 2021-07-07:
Thank you so much for the fast fix! We will watch your development closely :)

closed 2021-07-07 by skymaker-c2is