jvde-github/AIS-catcher #2: Cant open USB device (Win10, RTL-SDR V3 dongle)

obama, 2021-07-15:
when i try to test the application, i get the error
```
Device selected: ð­┌, , SN:
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
RTLSDR: cannot open device.
```
each time some other utf8 garbage is shown as name of the device.
i have uploaded the rtlsdr.dll which is needed for sdr#
i use the silver RTL-SDR V3 dongle.

aisdeco2 can find the device correctly.

jvde-github, 2021-07-15:
Hi, thanks for reaching out. 

Did you use the binary files from the release section, i.e. 1) unzip all files from AIS-catcher.v0.08.Win32.zip in one directory, 2) on the command prompt, change to that directory and start AIS-catcher?

Could you provide me with the output of the command "AIS-catcher -l"? 

The ZIP file has the equivalent of rtlsdr.dll in the form of librtlsdr.dll so no need to copy or upload any files. 

Thanks, Jasper

closed 2021-09-09 by jvde-github

obama, 2023-09-18:
what is wierd i just tried it today again without updating ais-catcher or anything and it works suddenly.. maybe windows updated the driver?