jvde-github/AIS-catcher #9: rtlsdr_set_bias_tee was not declared in this scope

unxs0, 2021-09-08:
ubuntu 16.04
had to add -std=gnu++11 to gcc in Makefile
Still have issues...

`gcc  -c Main.cpp IO.cpp DSP.cpp AIS.cpp Model.cpp Utilities.cpp Demod.cpp DeviceRTLSDR.cpp DeviceAIRSPYHF.cpp DeviceAIRSPY.cpp DeviceFileRAW.cpp DeviceFileWAV.cpp -O3 -Wno-psabi -ffast-math -DHASRTLSDR -std=gnu++11 
DeviceRTLSDR.cpp: In member function ‘void Device::RTLSDR::setBiasTee(int)’:
DeviceRTLSDR.cpp:103:33: error: ‘rtlsdr_set_bias_tee’ was not declared in this scope
   if (rtlsdr_set_bias_tee(dev, a) != 0) throw "RTLSDR: cannot set bias tee.";
                                 ^
Makefile:32: recipe for target 'lib-rtl' failed
make: *** [lib-rtl] Error 1
`

jvde-github, 2021-09-08:
Hi. I have added the -std=c++11 flag to the Makefile. Thanks for the suggestion. I did not test on Ubuntu 16.04.

The second issue might be related to the version of the rtlsdr library on Ubuntu 16.04 (0.5.3). One solution might be to upgrade the library in some way although this might be challenging. Another approach could be to just to comment out this particular line and forego this functionality. I have done that optionally in the latest version which can be activated on the command line as follows:

make CFLAGS=-DLIBRTLSDR_LEGACY rtl-only

Perhaps that helps?

Jasper

unxs0, 2021-09-08:
Summary: Works for me.

Thanks Jasper!

Worked 100%.

Your rtl ais is 10x faster than the others I have. Is there someway to validate results or develop a testing method to be able to compare to commercial products like the widely used in Marine Traffic, the SLR350Ni receiver.

Thanks!

jvde-github, 2021-09-08:
Thanks! What you mean with faster? In terms of message count?

The question on comparison crossed my mind as well but I have no idea how to test that. The comar receiver is a bit too expensive to buy just for that purpose. It would be interesting to see how close you can get with a cheap rtl dongle to professional equipment. Maybe I can contact a local marintraffic station in the area to see if they could help.

unxs0, 2021-09-08:
Welcome!

Yes messsage count, compared same antenna, same CPU, just different rtl-ais binary.

HAVE STATION

I have a station and it it is doing quite well (compared with the 2 other stations in my area) with your AIS-catcher running on a low power RPI 3B+:

https://www.marinetraffic.com/en/ais/details/stations/1839

I have a cheap dongle* running now but ***connected directly with no coax*** to a well tuned jpole antenna.
(*with some minor but important hardware and cooling mods)

Cheers!

unxs0, 2021-09-08:
Welcome!
Yes faster...vs. same cpu and antenna.
Other details privately.
Cheers!

jvde-github, 2021-09-09:
Sounds promising! Thanks for sharing.

closed 2021-09-09 by jvde-github