jvde-github/AIS-catcher #10: Possible Memory Leak when running for a long time?

McNugget6750, 2021-09-10:
I have to do more investigation on this but I noticed that every ship and every message gets catalogued and counted. I had AIS-Catcher running successfully without issues with OpenCPN on a Raspberry Pi 3B+ and after about 30h to 36h the entire system crashed. I saw messages on the AIS-Catcher terminal mentioning buffer-overflow or buffer-overrun. Nothing reacted, the mouse cursor was frozen. Every 10 seconds or so, there was some motion on the screen but not even enough to print an entire new line of text in the terminal.

I'll run the test again but I suspect some sort of memory leak in AIS-Catcher that's not there when using RTL_AIS and AIS-Decoder. Both of those run without issues for multiple days but perform severely worst overall. My range essentially doubles when using AIS-Catcher so I'm very interested in getting this into a stable state :)

jvde-github, 2021-09-10:
Hi. I am curious to hear if you can replicate the issue and get some more clues on the precise error message. 

In principle after the decoder is up and running there should not be any memory being allocated (i.e. messages are not stored, only counted), apart from some string manipulation to create the messages. I had tested for memory consumption over time but not on a RPI 3B+. 

Keen to hear if you can reproduce the issue and share some more details. I am wondering if this issue still occurs if we switch off UDP send out.

Thanks, Jasper

McNugget6750, 2021-09-10:
I started the test. Arguments are from memory but they made sense to me.

[image]
[image]
[image]

jvde-github, 2021-09-10:
Thanks. Curious to see what happens...

jvde-github, 2021-09-13:
Hi, any further insights from the test already?

McNugget6750, 2021-09-13:
Yeah, I had it running all week now. I’m not sure what changed but it’s rock solid at the moment. Same system, same config. Just rebooted and let it run for a week. I’ll close this for now and get back to you once this stability changes.
BTW: It takes only 45MB of RAM. It seems to run very well on RPI3B

Thanks for your great work! 
Cheers
Timo

closed 2021-09-13 by McNugget6750

jvde-github, 2021-09-13:
Thanks Timo. I did check separately for memory leaks and did some further thinking but have no idea what could cause instability apart from the higher system load compared to other s/w that could impact the system as a whole.

Thanks and keep me posted. Cheers, Jasper