jvde-github/AIS-catcher #11: Suppress "( MSG: ... MMSI: ...)" output

c-mcst, 2021-09-25:
Hi,

So far I am loving AIS-catcher. It is doing an impressive job of receiving messages compared to other libraries I've used. It would be great if there was an option to suppress the extra info after the AIVDM sentences (which is sometimes very useful) on output to increase interoperability with other software like chartplotters that require strictly the NMEA sentences as input. Maybe you've already implemented such a feature and I have just missed it, but I didn't see it in the documentation. 

Thanks so much for all your work on this.

jvde-github, 2021-09-25:
I am glad to hear! 

I have  added a switch (-n) that will display NMEA messages on screen without any details. Does this work for you? 

Your feedback made me think as well and perhaps it is better to remove the -q switch in future versions and only have something like -nx that activates NMEA on screen and controls the level of details via x.  

Thank you, Jasper

c-mcst, 2021-09-26:
Yes, that works perfectly. Using the -n switch I'm now able to feed directly to a number of other pieces of software over UDP. Thanks so much! 

That's an interesting idea to make the -q switch go away in favor of -nx. I think that would work very well for the crowd that wants to take the NMEA sentences and process them using other software, but it may be confusing for the beginner if there's no output to the screen by default. It's certainly worth thinking about. 

Cheers,
Chris

closed 2021-09-26 by c-mcst