jvde-github/AIS-catcher #13: Non-coherent models sometime better and MSG27

troosh, 2022-02-11:
Perhaps these two files might be interesting to explore (look to root of this [folder](https://drive.google.com/drive/folders/14EZC_yEmPqrTdSXzBB3F9aps1qND9ZM4)):
 
1) Non-coherent models receive more messages
 
RAW file Settings: -ga CU8 file 2021-08-14-10-54-29_rtlsdr_162000000Hz_1920000Sps.AGCoff_22.9db.iq
...
[AIS engine v0.34]:                      33 msgs at 1.2 msg/s
[AIS engine v0.34]:                      33 msgs at 1.2 msg/s
[Base (non-coherent)]:                   46 msgs at 1.6 msg/s
[Standard (non-coherent)]:               **50** msgs at 1.8 msg/s
 
2) Reception of MSG 27 for satellites at 156.8 MHz
...
RAW file Settings: -ga CU8 file 2021-08-15-17-29-43_rtlsdr_**1568**00000Hz_1920000Sps.iq
Generic settings: sample rate -s 1920K
!AIVDM,1,1,,B,KqNSLg1Bik=V3h,0*68 ( **MSG: 27**, REPEAT: 3, MMSI: 636017854)

jvde-github, 2022-02-11:
Thanks Alexander!  The default model with "-m 2 -go PS_EMA on" (was planning to make this the default) seems to find a few more messages (33->41) but still worse than the FM decoder. Very interesting to have a look at. Will let you know if anything interesting comes out of it.

jvde-github, 2022-02-18:
I extracted a problematic example ([example1.raw](https://drive.google.com/file/d/1hsxpgK4dnkG3H92V-4t8kXU4E7KbQ-2X/view?usp=sharing)) from the file and looked through my collection of recordings for other broadcasters for which the default model did perform worse than the FM decoder. Found one: a small class B vessel for which I extracted another sample ([example2](https://drive.google.com/file/d/1ve0ndqAsZspXMXiXNzxaUB3QQXV6HD2Z/view?usp=sharing)).

You can run them through different decoders as follows:
```
AIS-catcher -r CS16 example1.raw -s 96000 -m 0 -m 2 -m 4 -v
```

Interestingly, one way to avoid this problem is to use shorter averages to determine the phase offset. I have implemented that in model ``-m 4``. Results for this model are also added to the table in the previous post.

This model (```-m 4```) on my collection of recordings find 33% more messages that the plain vanilla FM decoder. The default model actually sits at 40%. The slight increase in robustness for frequency instability comes at a cost of general loss of sensitivity. The trick is to combine the two somehow. Experimentally it seems though that any additional complexity will come at a cost in general reception.

closed 2022-11-20 by jvde-github