jvde-github/AIS-catcher #14: Suggestion: list supported sample rates in the readme

dziban303, 2022-02-21:
I was trying out the -F switch on a Pi Zero W, but I kept getting buffer overrun warnings at 1536Msamples. Wanting to decrease the samplerate to find one which didn't throw warnings, I found that nearly every value I gave it would present "sample rate not supported by the model." It took far too long for me to realize I could probably find the supported rates in the code, and indeed, [Model.cpp](https://github.com/jvde-github/AIS-catcher/blob/0abdd120610ea1dcde3f919cb3eb0870a317de25/DSP/Model.cpp) seems to have it.

It's possible I'm the only fool who's had this problem, but if not, perhaps it'd be a good idea to list what sample rates AIS-catcher will accept somewhere in the readme?

jvde-github, 2022-02-21:
Hi, I have added the sample rates to the README and also added the ability to specify sample rates as ```-s 1536K``` instead of ```-s 1536000```. Initially my plan was to only provide for one or two sample rates, but this has grown since to support more hardware and experimentation. I think it is a good addition to the README to list them. I hope this helps.

Thanks for the note.  I had a similar observation on the RPI1 compared to the RPI4, despite managing to get the CPU usage below <60%. I was keen to understand whether performance was also lacking because of other things (like power supply and/or shielding) but did not have time to experiment with this yet.

jvde-github, 2022-02-22:
Thanks again for the suggestion. I will close this issue. Thanks!

closed 2022-02-22 by jvde-github

dziban303, 2022-02-22:
Sorry, meant to reply when I got home, and subsequently forgot. Thank you for the addition, and the software generally! AIS-catcher is excellent, it solved a lot of problems for me.

jvde-github, 2022-02-22:
Thank you, great to hear!