jvde-github/AIS-catcher #16: UPD connection issues

guidocioni, 2022-04-21:
I'm running `AIS-catcher` on a Raspberry Pi, broadcasting messages over UDP, to be received on a laptop with Open CPN. 

On the raspberry pi side, where the dongle is connected, I'm running `AIS-catcher` like so
```
AIS-catcher -gm lna AUTO vga 12 mixer 12 -v 10 -d:0 -u 0.0.0.0 12345
```
everything works fine and I get messages printout. 

```
AIS-catcher (build Apr 21 2022) v0.35
(C) Copyright 2021-2022 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Device selected: AIRSPY, AIRSPY, SN: B58069DC391F5913
Airspy Settings: -gm lna AUTO mixer 12 vga 12 biastee OFF
Generic settings: sample rate -s 3000K
!AIVDM,1,1,,B,139t@WP000PeLRNN`<9v49RN0L5P,0*2D ( MSG: 1, REPEAT: 0, MMSI: 211751070)
!AIVDM,1,1,,B,139udkPP06PeVtHN`KdIhOvNR<4:,0*0A ( MSG: 1, REPEAT: 0, MMSI: 211774670)
!AIVDM,1,1,,A,139S:k0P00PeTAjN`>8v4?vN20R;,0*33 ( MSG: 1, REPEAT: 0, MMSI: 211339980)
!AIVDM,1,1,,A,139QeeP01fPe4gRN`opSG2TH00Sb,0*75 ( MSG: 1, REPEAT: 0, MMSI: 211316150)
!AIVDM,1,1,,A,139QoB0P1=Pe=uFN`af@UwvN00Sf,0*70 ( MSG: 1, REPEAT: 0, MMSI: 211318600)
!AIVDM,1,1,,A,339f8r0P00PeWlDNWoJP0?vN2C83,0*38 ( MSG: 3, REPEAT: 0, MMSI: 211519720)
!AIVDM,1,1,,A,33aN1t@P@8PeEtBN`N>V?@4L00oA,0*43 ( MSG: 3, REPEAT: 0, MMSI: 244810225)
!AIVDM,1,1,,B,139c?8?000PeLURN`=2rE48H0`9A,0*71 ( MSG: 1, REPEAT: 0, MMSI: 211472160)
!AIVDM,1,1,,B,139fa8PP00PeLr@N`5b=QOvN2@9L,0*0F ( MSG: 1, REPEAT: 0, MMSI: 211527970)
^C!AIVDM,1,1,,A,139G`r0P1VPeS1>N`li:CwvP25p8,0*6A ( MSG: 1, REPEAT: 0, MMSI: 211151080)
```

However, if I look inside `lsof` I can see that another port is used for `UDP`,
```
AIS-catch 218085          ubuntu   10u  IPv4 17411663      0t0  UDP *:51367
```
the port is randomly chosen every time that I start and it never matches my input. 

Furthermore, I cannot connect from Open CPN.
I used these settings, by matching the open port every time as it changes, but it never receives a package 

[image]

`192.168.178.36` is the local IP address of the Raspberry Pi. 

I tried to ping the port with `telnet 192.168.178.36 51367` and it does not find an open connection.

What am I doing wrong?

guidocioni, 2022-04-21:
Didn't manage to connect with Open CPN but installed AIS dispatcher and it's working fine

closed 2022-04-21 by guidocioni

jvde-github, 2022-04-22:
Hi Guido, thanks for raising the issue and reporting the fix. Your original issue could be that you are sending the messages to 0.0.0.0 which is the address of the raspberry and not the laptop with OpenCPN. A possible fix could be to use the IP address of the laptop on the AIS-catcher command line and use the same IP address in the OpenCPN setting (i.e. use the destination address instead of the source address). The port in OpenCPN should then be 12345.

Thanks for trying AIS-catcher!

guidocioni, 2022-04-22:
uhm, shouldn't the address be the one of the local machine where ais catcher is running?
I thought by exposing that over UDP then any device in the local lan would be able to access it.

jvde-github, 2022-04-22:
OpenCPN need to be in listening mode on a specific port. The IP address is less relevant and can therefore be 0.0.0.0 in the OpenCPN settings. AIS-catcher sends any message to the provided destination IP and port. 

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:options_setting:connections

guidocioni, 2022-04-22:
> OpenCPN need to be in listening mode on a specific port. The IP address is less relevant and can therefore be 0.0.0.0 in the OpenCPN settings. AIS-catcher sends any message to the provided [...]

wow, ok, that explains it. thanks for the clarification.