jvde-github/AIS-catcher #17: RTLSDR device

LC1C, 2022-05-26:
Hi, 
I don't know if there an issue or it somewhere with missmatch configuration.
Command AIS-catcher -l don't see any RTLSDR dongle. Drivers for RTLSDR are installed on PI ubuntu Mate 18.04.

For example there no problems with RSP1A it detects and device are in found list.
#AIS-catcher -l
Found 1 device(s):
0: SDRPLAY, RSP1A, SN: 180200XXXX

# SoapySDRUtil --find
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U 
  manufacturer = Realtek
  product = RTL2838UHIDIR
  tuner = Rafael Micro R820T

Found device 1
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 180200XXXX
  serial = 180200XXXX

Where it can be problem that AIS-catcher do not find RTLSDR device?

Thank you in advance

LC1C, 2022-05-26:
Thank you for answer!

there is output 

AIS-catcher -L
AIS-catcher (build Feb 17 2022) v0.34
Supported SDR(s): SDRPLAY RTLTCP

seems missing RTLSDR

jvde-github, 2022-05-26:
Great! The cmake file was relying on the input of pkg-config solely and I have now added some more additional directories to look for the library. Many thanks for your input!

closed 2022-05-26 by jvde-github