jvde-github/AIS-catcher #20: When building, it seems to think there's no support for biastee with librtlsdr v0.7.0-187-g710d

dziban303, 2022-06-05:
I tried to build again on a Pi 4 after pulling the latest from github, and saw this after doing `cmake ..`:
```make
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version v0.7.0-187-g710d
-- RTLSDR: found - /usr/local/include, /usr/local/lib/librtlsdr.so
-- RTLSDR: no support for biastee
```
It should support biastee. I also built it on a different Pi (Zero 2) with librtlsdr v0.6.0 and it worked fine:
```make
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0
-- RTLSDR: found - /usr/include, /usr/local/lib/librtlsdr.so
-- RTLSDR: support for biastee
```

`uname -a` on both systems: 
`Linux zero2 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`

jvde-github, 2022-06-05:
I changed a few things in the latest cmake files for the bias tee. Previously it was assumed it was available unless the version of librtlsdr was below 0.6. This gave build issues if the version number for rtlsdr could not be detected and was 0.5.3.  

I decided to turn it around and exclude bias tee support except if we can establish a rtlsdr version of 0.6 or higher. I think this is cleaner.
It seems from the above that on the first machine cmake cannot determine the rtlsdr version number (from findpkg) but still can find the library by searching for the files and therefore does include rtlsdr support but without bias tee functions.

We can add a option to the cmake file to force incusion of bias tee support. There is also a cmake function that checks for the existence of particular library functions. Will have a look at it.

Thanks for flagging.

dziban303, 2022-06-05:
Yep! 
```make
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version v0.7.0-187-g710d
-- RTLSDR: found - /usr/local/include, /usr/local/lib/librtlsdr.so
-- Looking for rtlsdr_set_bias_tee
-- Looking for rtlsdr_set_bias_tee - found
-- RTLSDR: bias-tee support included.
-- Looking for rtlsdr_set_tuner_bandwidth
-- Looking for rtlsdr_set_tuner_bandwidth - found
-- RTLSDR: tuner badwidth support included.
```
I meant to say last night that I think it failed because of the extra stuff at the end of the librtlsdr version string; if it weren't there, presumably the greater-than check would've worked. 

For what it's worth, I believe the build of librtlsdr I'm using is [https://github.com/librtlsdr/librtlsdr](https://github.com/librtlsdr/librtlsdr), which ostensibly is version 0.8. I'm not sure why the version comes back with `v0.7.0-187-g710d`. Anyway thank you!

closed 2022-06-05 by dziban303