jvde-github/AIS-catcher #22: Help4newbie: Connecting to a SDR public feed

alsak0de, 2022-06-16:
hiya, would you please help me with the syntax to fetch from an actual SDR remote signal like the ones listed on https://airspy.com/directory/ ??

I've tried   AIS-catcher -t 31.31.235.80  5555 -s 240000 -v as an example but seems the received feed is different and AIS-catcher -t requires a RTLTCP feed that I suppose it is different.

Is there a way to inject a sdr:// feed instead of RTLTCP? If not, are there RTLTCP public servers to get the signal from?

thanks in advance,

jvde-github, 2022-06-16:
Hi, it is an interesting idea, you need to find a feed that receives AIS signals from vessels and allows to receive on 162 MHz. This reduces the options quite a bit, but there seem to are a few.
An easy way to check this is to use [SDR#](https://airspy.com/download/) and look at AIS signals around 162 MHz. I managed to found at least one feed which looks to have AIS signals (around Gothenborg).

Then the second step is to decode these messages. Unfortunately, AIS-catcher has the required protocol not implemented. You could record the signal perhaps in SDR# and the decode it in AIS-catcher but I cannot try that at the moment.  

Let me have a look how difficult it is to add airspy server support by adding an extra protocol (currently there is RTLTCP and NONE). 

Jasper

jvde-github, 2022-06-16:
Turns out that there is a [SoapySDR driver for spyserver](https://github.com/pothosware/SoapySpyServer) which has been recently published. So, installing this driver and using the experimental support for SoapySDR and SOXR downsampling, we could theoretically decode these signals, with commands like:

```
AIS-catcher -d SOAPYSDR -gu device "driver=spyserver,host=143.178.161.73,port=5556" PROBE on AGC off  -s 625000 -go SOXR on -v
```
Notice that the sample rate you need to provide and needs to be supported by the feed, which can be found by:
```
SoapySDRUtil --probe="driver=spyserver,host=143.178.161.73,port=5556"
```

Unfortunately, this does not work yet :-(, the device opens but the streaming stopped immediately in my test. This is likely an issue with AIS-catcher that I need to fix before release.

alsak0de, 2022-06-17:
Thanks!! Recording and then passing catcher works but the objective would be to get a RTMP type of audio stream that the catcher can be connected to.

jvde-github, 2022-06-17:
Hi, made a small fix and commands like this should work now:
```
AIS-catcher -d SOAPYSDR -gu device "driver=spyserver,host=143.178.161.73,port=5555"  -s 192K  -go SOXR on -v
```

jvde-github, 2022-06-18:
There is also another route that you could explore. I managed to get some AIS messages with this approach, see below (unfortunately not with the soapysdr driver) . The idea is to use a spyserver client and pipe the input into AIS-catcher:
```console
./ss_client iq - -r 143.178.161.73 -q 5555 -f 162000000 -s 192000 -g 12 |AIS-catcher -s 192K -r CS16 . 
```
ss_client is available [here](https://github.com/miweber67/spyserver_client) and I installed libsamplerate_dev
```
sudo apt install libsamplerate-dev as dependency:
```
Output:
```
./ss_client iq - -r 143.178.161.73 -q 5555 -f 162000000 -s 192000 -g 12 |AIS-catcher -s 192K -r CS16 . 
iq filename: -
bins for res: 1920   fft bins: 2048   resolution: 93.75Hz
SS_client_if(143.178.161.73, 5555)
SS_client_if: Trying to connect
AIS-catcher (build Jun 16 2022) v0.35
(C) Copyright 2021-2022 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
SS_client_if: Connected

**********
Device Info:
   Type:                 2
   Serial:               825307186
   MaximumSampleRate:    768000
   MaximumBandwidth:     660000
   DecimationStageCount: 8
   GainStageCount:       0
   MaximumGainIndex:     0
   MinimumFrequency:     0
   MaximumFrequency:     1700000000
   Resolution:           16
   MinimumIQDecimation:  2
   ForcedIQFormat:       0

**********
Client sync:
   Control:     Yes
   gain:        12
   dev_ctr:     330000
   ch_ctr:      330000
   iq_ctr:      330000
   fft_ctr:     330000
   min_iq_ctr:  82500
   max_iq_ctr:  1699917500
   min_fft_ctr: 330000
   max_fft_ctr: 1699670000

**********
Client sync:
   Control:     Yes
   gain:        12
   dev_ctr:     330000
   ch_ctr:      330000
   iq_ctr:      330000
   fft_ctr:     330000
   min_iq_ctr:  82500
   max_iq_ctr:  1699917500
   min_fft_ctr: 330000
   max_fft_ctr: 1699670000
SS_client_if: Ready
Exact decimation match
Desired decimation stage: 2 (768000 / 4 = 192000) resample ratio: 1
ss_client: setting center_freq to 1.62e+08
SS_client_if: Starting Streaming
SS_client_if: Lost 1 frames from SpyServer!
!AIVDM,1,1,,A,13aPNrPP00PJp1lMbihh0?vb2@<E,0*48 ( MSG: 1, REPEAT: 0, MMSI: 244850410)
!AIVDM,1,1,,A,13`p=J@P00PJon:MbiP>4?vb20Rc,0*76 ( MSG: 1, REPEAT: 0, MMSI: 244190569)
!AIVDM,1,1,,A,33aDoQ000jPJqk@Mbs3AoAJT05US,0*3A ( MSG: 3, REPEAT: 0, MMSI: 244660100)
!AIVDM,1,1,,B,13aEQ7wP00PJou6MbiS@0?vd20S5,0*58 ( MSG: 1, REPEAT: 0, MMSI: 244670751)
```

jvde-github, 2022-06-26:
Ok, I added some initial SpyServer support which can be activated with a comment like:
```
AIS-catcher  -y 89.245.170.117 5555 -gy GAIN 14 -v
```
There are still a few things to clean up (error recovery, more stream formats, etc), but it seems to work:
```
AIS-catcher (build Jun 26 2022) v0.35
(C) Copyright 2021-2022 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Connecting to SpyServer...
Device info:
  Serial: 942682164 MaximumSampleRate : 768000 MaximumBandwidth : 660000
  DecimationStageCount: 8 GainStageCount: 0 MaximumGainIndex: 0
  Minimum/Maximum Frequency: 0/275000000 resolution: 16
  MinimumIQDecimation: 2 ForcedIQFormat: 0
Client:
  CanControl: 1 Gain : 0 DeviceCenterFrequency : 89600000
  IQCenterFrequency: 89600000  Minimum/Maximum Frequency: 82500/274917500 resolution: 16
SPYSERVER settings: -gy host 89.245.170.117 port 5555 gain 14
Generic settings: sample rate -s 192K  
Client:
  CanControl: 1 Gain : 0 DeviceCenterFrequency : 89600000
  IQCenterFrequency: 89600000  Minimum/Maximum Frequency: 82500/274917500 resolution: 16
[AIS engine v0.35]:                      0 msgs at 0 msg/s
!AIVDM,1,1,,B,4025btl47wk?0<tSF0l4Q@000l5I,0*48 ( MSG: 4, REPEAT: 0, MMSI: 2190067)
[AIS engine v0.35]:                      1 msgs at 0.2 msg/s
!AIVDM,1,1,,A,4025btl47wk?1<tSF0l4Q@000PS:,0*63 ( MSG: 4, REPEAT: 0, MMSI: 2190067)
[AIS engine v0.35]:                      2 msgs at 0.2 msg/s
!AIVDM,1,1,,B,4025btl47wk?0<tSF0l4Q@000d5Q,0*58 ( MSG: 4, REPEAT: 0, MMSI: 2190067)
[AIS engine v0.35]:                      3 msgs at 0.3 msg/s
^C----------------------
[AIS engine v0.35]:                      3 msgs at 0.1 msg/s
```

jvde-github, 2022-06-29:
As Spyserver support is now included I propose to close this issue.

closed 2022-06-29 by jvde-github