jvde-github/AIS-catcher #23: Received signal strength / SNR

anttinousiainen, 2022-06-27:
Great software, works really well onboard my liveaboard boat as a backup to a "real" AIS transmitter/receiver, thanks for this!

Anyways, I have been wondering, would it be possible and/or worth implementing to give even an estimate on the received signal strength / SNR for each received message? This is something not found on any receiving software I know of, and would be quite helpful in troubleshooting problems in AIS transmission / reception - which mostly are really problems in the cabling and/or antenna issues.

anttinousiainen, 2022-06-27:
"Out of curiosity, does your set up give an idea of how SDR AIS decoders compare to real AIS decoders?"

A bit hard to say yet, as I have a setup where my Furuno FA150 transponder has a dedicated proper AIS-frequency matched antenna high up and my Icom m605 VHF/AIS -combo also has a very good long whip antenna also well placed and high up - these two came with the boat- and my SDR has a crappy small sort of dipole thingie :)

So yes, I plan to install a better proper marine VHF-antenna for my SDR radio and also maybe get a better, more sensitive SDR receiver, then I can do some real meaningful comparison between them, but empirically there seems to exist a huge variance among the AIS installations. Some ships you can receive like 30nm away, some smaller boats (with not so good antenna installation I guess) can only be "heard" less than a mile away.

If there was an actual number on the received signal strength, it would be very helpful to notify fellow boaters on their potentially weak AIS transmission with some "hard data" to back it up.

jvde-github, 2022-06-29:
I like the idea and have added it to the TO DO list. Might also be interesting in comparing different SDR hardware.

Thanks for raising.

anttinousiainen, 2022-06-29:
Great news, thanks!

I'll keep you updated once I upgrade my SDR-setup and antennas. Now I did some quick trials, my ICOM with a good antenna seems to receive roughly double the AIS frames compared to my SDR with not so good one.

I have a feeling they'll be more even with equal antennas.

sm3ulc, 2022-07-02:
> Thanks! Need to think a little bit about this but definitely something that can be explored if it does not make the decoder too complicated.
> 
> There is a FFT before the decoder that corrects for frequency shift, perhaps that information can be used somehow to report some relative strength (and frequency shift) by passing information downstream through the decoders as meta data.
> 
> Out of curiosity, does your set up give an idea of how SDR AIS decoders compare to real AIS decoders?
> 
> Thanks for sharing your thought. Kind regards, Jasper

Hello! I can provide data in a few weeks. I have a hardware decoder and antenna splitter, RTLs and SDRPlay that I can run in parallell.

anttinousiainen, 2022-07-14:
Actually, it might make sense to add other "receiving metadata" as well, if feasible and/or reasonable.

In addition to signal strength, accurate-ish receiving timestamp would be very interesting addition, as well as estimated frequency offset from nominal.

But these are more of "interesting" additions, the received signal strength I say would be a priority over these.

jvde-github, 2022-07-14:
Like the thinking. On the list... It would be useful if the output is such that it can be used downstream. Any thoughts on what a convenient way is to output this data for further processing?

anttinousiainen, 2022-07-14:
"Any thoughts on what a convenient way is to output this data for further processing"

--format-all-output-in-json -option?

anttinousiainen, 2022-07-15:
Speaking of options, for testing and tuning, timeout option/parameter would be _very_ handy:

-t / --timeout n sec, decode for n secs and terminate

I use an external timeout command to run for a certain time and then check running with ppm +-1 to see if "better" tuning is available (cheap SDRs do drift, they just do), but it is not very readily available under Windows, so internal parameter would be much better.

anttinousiainen, 2022-07-15:
Also on a same note, should it be possible to do the ppm-adjustment to a stored iq-file and/or one read from stdin? I seem to not manage this if I try.

Would make formerly mentioned "autotuning" easier, as could test multiple ppm-settings on the same iq-data in parallel.

jvde-github, 2022-07-15:
> 

I have implemented something last night to show signal levels. Need some testing and finetuning. Will post in a bit. This one I can also quickly add.

jvde-github, 2022-07-15:
On PPM adjustment on files, I don't think will implement that soon as there are 3rd party tools for that (at least on Linux).

CSDR is quite nice for that: https://github.com/ha7ilm/csdr

You can for example take a file, shift the frequency a bit and do AIS-decoding with the following command:

```
cat  ~/Signals/rtl/posterholt_1536_1.raw | csdr convert_u8_f | csdr shift_addition_cc -0.0003 | ./AIS-catcher -r CF32 . -b -v
```

jvde-github, 2022-07-15:
Hi! On your other request for a timeout, was fairly easy to add (and found it useful myself as well).
It only works if you run via a SDR device (not from file) and timeout can be activated using the following command (e.g. 10 seconds):
```
AIS-catcher -T 10
```

Hope this helps as well :)

anttinousiainen, 2022-07-16:
Super! Works really well, huge thanks!

Though I noticed a slight regression, either in the code or my equipment - or my location. This new version seems to heavily favor reception on channel A. Now roughly 95%+ of the messages are on channel A, though some B messages make it through from time to time.

Could be also local interference.

jvde-github, 2022-07-16:
Great! Good additions so thanks for suggesting. 

But really odd that channel B has disappeared practically. If you run v0.35 or v0.36, do you have the same absence of Channel B messages?

The decoding engine did not change in the last few versions and I am doing regression testing on a large set of recordings before every release (total number of messages identified remained constant). So am a bit puzzled. Let me check.

anttinousiainen, 2022-07-16:
It seems to be identical on v0.36 and v0.37, so it must be some local issue - which is weird as my Icom shows constant 50/50 on A and B. I am 99% certain v0.36 showed better distribution when I was moored in an another location a while back.

I ordered a bunch of new SDRs, antenna cables and antennas, once they arrive I do some more thorough testing.

jvde-github, 2022-07-16:
Yes, keep me posted if you find anything with the software. Can also have a look with a program like SDR# at the signal or also happy if you email me a short recording (using ``rtl-sdr``) so we can have a look.

jvde-github, 2022-07-16:
Was planning to write a small variant on AISdispatcher that allow me to do some statistics on different input channels (to compare decoders, hardware for completeness, range, etc) and display some more statistics and database dumping. Was planning to add it there but that will take some time and you want to have the new info in a user program indeed like OpenCPN. 

This additional data is fairly trivial to add to JSON output but was reading that AIS meta data is not really evolved: https://gpsd.gitlab.io/gpsd/AIVDM.html#_u_s_coast_guard_extended_aivdm

I can simply add things but I think this requires some bigger changes to OpenCPN et al.

Perhaps others have any ideas on standards and options?

anttinousiainen, 2022-07-16:
Actually that Coast Guard NMEA extension metadata would be a good option, I think. No need to invent any new things, as it has all the required fields defined, yes?

Then would only need to add support for said extension to eg. OpenCPN?

jvde-github, 2022-07-16:
Indeed, OpenCPN needs to define a tag block that allow us to pass on this information. Then we should be able to pass on that info.

The other solution, which is not as neat but probably the easiest route, is to write a short script in Python or so that takes the output of the decoder and tracks and displays the signal level for each MMSI.

jvde-github, 2022-07-17:
I have added a new switch ```-o 3``` to the program that formats the output such that the NMEA lines and meta data is embedded in a JSON string which would make it easy for a 3rd party program to read and decode, e.g. compare signal level with nearby vessels or so. Perhaps a OpenCPN plugin :-)

jvde-github, 2022-07-23:
Since the signal power calculations (``-M D``), timeout (``-T xx``) and timestamping (``-M T``) are implemented I suggest we close out this issue.

Thanks, Jasper

closed 2022-07-23 by jvde-github