jvde-github/AIS-catcher #24: Question: Anybody tried sending data directly to MatineTraffic

soend, 2022-06-29:
Hello,

Good job on the awesome tool, it works great with OpenCPN.
I was wondering has anybody tried to send NMEA sentence directly from AIS-catcher to MarineTraffic?

I did try it with my setup but MarineTraffic always shows station offline and 0 messages received.

jvde-github, 2022-06-29:
Hi, I use AIS dispatcher in my setup which seems to work. To understand the issue, have you managed to send to MarineTraffic via other routes and/or other setups (e.g. software)?

jvde-github, 2022-06-29:
I have tried on Windows as well with the command I provided above. It took a one or two minutes before I saw the first activity in the "AIS messages per minute (real time)" graph. And it is now running for 15 minutes and still says "ONLINE - NO DATA". I think this can take quite some time - site suggests that statistics are only updated hourly.

Silly question, but you are using the correct port number? E.g. something like station number + 5320?

anttinousiainen, 2022-06-29:
I have been using this tiny python script for ages to send to Marinetraffic, works both direct from a "real" AIS receiver and SDR-one. It expects NMEA/AIS messages into sdtin and sends to MT:

'''
import fileinput
import sys
import socket

UDP_IP_M = "5.9.207.224"
UDP_PORT_M = 11998

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM) 

for line in fileinput.input():
        MESSAGE = line.rstrip().split(" ")[0]
        sock.sendto(MESSAGE, (UDP_IP_M, UDP_PORT_M))
'''

then I run it reading from fifo like this:

'''
cat /dev/shm/nmea.fifo | pv -b -t -r -a -i 2 -l | python marinetraffic_udp.py
'''

soend, 2022-06-30:
Thanks for all your feedback. 
It seems like i was just too inpacient, and expected it to work straight after creating new station in MarineTraffic. Today morning when tried again sending data my station came online and is working with sending data directly from AIS-catcher.

closed 2022-06-30 by soend

jvde-github, 2022-06-30:
Good to hear, thanks both!