jvde-github/AIS-catcher #25: Question/feature/future - virtual stacking

sm3ulc, 2022-06-29:
Awesome development of AIS-decoder! A thought for future. Today a very good colinear or similar AIS-antenna is rare and expensive. A interesting field to explore would be to combine signals from 2+ receivers into one. For example, to 2x RTLs need to share clock or is it enough with the RTLv3, or even the black/blue ones?

jvde-github, 2022-06-30:
Thanks for the suggestion. Do you have an example, description or paper how that would work? Of course you can run multiple instances of AIS-catcher on one PC but that is probably not what you mean. Thanks Jasper

sm3ulc, 2022-07-02:
No expert in signal theory so bare with me.. :)  Basically ( rtl1_I[n] + rtl2_I[n+x] ) / 2, n - number of samples, x - phase shift, 1 to some reasonable limit. This before the decimation & filtering. 

If streams decimated & filtered in parallell, then  rtl1_I[n] + rtl2_I[n] should be enough since the bitrate is only 9600?

jvde-github, 2022-07-03:
Thanks, was planning to play a bit with multiple SDRs anyway, so will take this thought into account. Thanks again.

jvde-github, 2022-11-20:
It is on the to-do list.. Might take a while though.

closed 2022-11-20 by jvde-github