jvde-github/AIS-catcher #27: SDRPLAY: API v3.x not running

pingel20, 2022-07-14:
Hi Jasper,

Starting the x64 version of AIS-catcher v0.37 gives me: 'SDRPLAY: API v3.x not running'
and quits.

I use a RTL-SDR dongle.
v0.36 runs just fine.

Any ideas?

jvde-github, 2022-07-14:
Hi!

Oops.... I included SDRPlay support by default but did not test properly on a PC without the API installed. 

As a quick fix I will provide a separate executable without SDRPlay support to avoid some unexpected behaviour of the program in case the API service is not running. Can you test of the new executable works? 

Thanks for flagging, would not have picked this up myself.

jvde-github, 2022-07-15:
Thanks for bringing up the issue! (prettig weekend :-) )

closed 2022-07-15 by jvde-github