jvde-github/AIS-catcher #28: Leading zero not recognised for Airspy serial number.

gm8arv, 2022-07-17:
I have an Airspy mini with a serial number like: 04A464C8348A8C0C.  AIS-catcher doesn't recognise this in the -d parameter, instead requiring: 4A464C8348A8C0C, missing the leading zero.  Only tested with the Airspy mini.  Windows-10/32, v0.37.

Otherwise a first-class program!  Thanks for making it available.

jvde-github, 2022-07-17:
Thanks for flagging and uploaded a fix that changes airpsy serial codes to 16 character strings including any leading zeros.
Thanks for raising, makes more sense to change it indeed.

Interestingly, it seems that the airspy_rx tool is translating the serial number to  **0x**04A464C8348A8C0C.

jvde-github, 2022-07-18:
Hi David, I propose to close this item,;the fix is in the latest development version and will be part of the next release. Many thanks for flagging, would never have catched this.

Thanks and regards, Jasper

closed 2022-07-18 by jvde-github