jvde-github/AIS-catcher #31: Reordering -h output

dziban303, 2022-10-31:
I think there may be benefit to rearranging the `-h` help output. Many programs have the switches in their help screen sorted alphabetically, which I think would be an improvement on the current arrangement since the order of options doesn't matter:

[image]

Keeping each of the 'sections' as-is would be fine, but within the sections, alphabetizing would make it easier to scan when looking for a specific option.

What's your take?

jvde-github, 2022-11-01:
Like it, will do, thanks!

jvde-github, 2022-11-01:
Done, hope it is now a bit easier to use. Thanks.

dziban303, 2022-11-01:
Much better, thank you! Now back to troubleshooting the service performance issue I was having...

closed 2022-11-01 by dziban303