jvde-github/AIS-catcher #32: Poor performance running as a systemd service

dziban303, 2022-11-01:
I tried installing a systemd unit file using [the script by @abcd567a](https://github.com/abcd567a/install-aiscatcher) but for some reason the decoding performance is on the order of _30%_ of what I get when running AIS-catcher from bash, or using `screen`, or spawning at boot via `rc.local`. 

Launched via `screen` with -v set to 10 seconds, ran for 1760 seconds, so about 3.1 messages per second.
 * `[AIS engine v0.38]    received: 30 msgs, total: 5447 msgs, rate: 2.9946 msg/s`

Launched via systemd, same settings used, ran for 1793 seconds, for 0.89 messages per second.
`[AIS engine v0.38]    received: 8 msgs, total: 1589 msgs, rate: 0.796202 msg/s`

The processes use the same amount of CPU cycles; `htop` reports it as 28-30% regardless of how AIS-catcher is launched. I'm running this on a Raspberry Pi 3B+ running buster.

Any ideas as to why this would be happening?

jvde-github, 2022-11-01:
Did you change the settings for the systemd using:
```
 sudo nano /usr/share/aiscatcher/aiscatcher.conf 
```
I would remove everything except for the ``-v``. The script is great but the standard settings in this file will result in poor performance. Is that it?

dziban303, 2022-11-08:
I'd initially used the same settings I've been using, successfully, when running it manually. I tried eliminating options from my ais-catcher config, I tried tweaking settings in my systemctl unit file, nothing seemed to make any improvement. I tried it with v0.39 as well. So I admitted defeat and have just gone back to spawning it in a `screen` instance via `rc.local`. It must be some peculiarity with my setup.

closed 2022-11-08 by dziban303