jvde-github/AIS-catcher #34: RTLSDR Buffer overrun

jnbrnk, 2022-11-09:
Iḿ getting RTLSDR Buffer overrun every minute when running :
AIS-catcher -a 192K -u 127.0.0.1 2233 -q -gr RTLAGC on TUNER auto

When using -F parameter on Pi 3B as advised on low power systems I get : Model: unknown setting

jvde-github, 2022-11-09:
Thanks, I uploaded a bug fix and should be fixed. Apologies and thanks for flagging.

AIS-catcher does not require much resources so should run without problem on a 3B unless there are (temporarily) a lot of other processes. A better way to reduce the resource requirement of the program is to run on a lower sampling rate:

 AIS-catcher -a 192K -u 127.0.0.1 2233 -q -gr RTLAGC on TUNER auto -s 288K

jvde-github, 2022-11-09:
Hope that solves it. In the mean time I quickly added a function that would allow to increase the buffer count:
```
 AIS-catcher -a 192K -u 127.0.0.1 2233 -q -gr RTLAGC on TUNER auto BUFFER_COUNT 12
```
Default buffer count is 2 so that gives some more slack to recover from a temporary slow process.

jnbrnk, 2022-11-10:
Built the new version. Tested with extra buffers. Result is only occasional overrun, processor load (htop) 85% when receiving/decoding an area with abt 75 vessels

jvde-github, 2022-11-10:
Great, thanks. Out of curiosity, is this load only from AIS-catcher? Even on an early version of the RPi I have not seen it above 50% with -F.

Sounds like it has been resolved and we can close this issue? Thanks, Jasper

jnbrnk, 2022-11-11:
Pi is running Aisdispatcher and ais-catcher. itś a model B rev 2.
Hardware	: BCM2835
Revision	: 000e
Serial		: 00000000875a0d67
Model		: Raspberry Pi Model B Rev 2

All fine now so closing this issue. Keep up the good work on ais-catcher ! Will keep on testing. 

My other station https://www.marinetraffic.com/en/ais/details/stations/17133 is also running ais-catcher, on a pi zero-W

closed 2022-11-11 by jnbrnk