jvde-github/AIS-catcher #39: RTL-SDR sample rates

otm64, 2022-11-24:
Due to my local conditions I'd like to try a slightly lower sample rate for the RTL-SDR. However, when I try to use 768K an error is thrown about an invalid rate. Same problem with 384 and 576. 288, 1152, 1536 and 2304 work OK. Am I missing something? I thought all sample rates listed on the README would be supported by the RTL-SDR.

jvde-github, 2022-11-24:
According to the documentation, the RTL-SDR hardware supports the following rates:
```
possible values are:
   225001 - 300000 Hz
   900001 - 3200000 Hz
   Sample loss is to be expected for rates > 2400000
 ```
The AIS-catcher software supports all sample rates above 96K (because it also supports the Airspy HF+ etc) and whatever rate the user sets, this will be passed on to the hardware. The RTL-SDR will return an error outside of the above two ranges which is what you experienced.

The listed sample rates in the README are the rates at which the software runs most efficiently. So it makes sense to pick one of the numbers from this list. For the RTL-SDR that means that 288K and 1536K are logical choices. Having said that you can also run at 225001 (but not 225000 or 768000 as these are not supported by the dongles).

Having a look again at the README on this topic, suggests it can be a bit clearer what the takeway for the user is. Will reword it a bit. Hope that helps.

otm64, 2022-11-24:
Thanks - I knew all that about the gap in RTL-SDR rates at one time and regrettably forgot it. Sigh. Great project you have going, keep up the excellent work.

closed 2022-11-24 by otm64