jvde-github/AIS-catcher #45: Webserver doesn’t work behind reverse web proxy

kx1t, 2022-12-04:
Hey, this probably very premature because you’re still developing this, but your web server doesn’t appear to receive data if deployed behind a (nginx) based reverse web proxy. 

In my case: https://kx1t.com/ais -> http://10.244.136.18:90 (AIS-catcher target device with argument “-N 90”)

This is what it looks like when I go directly http://10.244.136.18:90
[image]

This is what it looks like from the reverse proxied address:
[image]

I’ve been on my mobile and can’t quite debug this, but I suspect that your data stream (websocket?) uses an absolute directory name (`/xxx/yyy`) instead of a relative one. Chrome inspector should be able to tell for sure.

jvde-github, 2022-12-04:
Thanks for trying and any feedback at any stage is appreciated! I like what you are doing btw, maybe will copy this set up some day so I can monitor my station when I am not on my network. 

The code is super trivial now: the statistics are made available in a json file that the HTML page pulls from the server.
Quick test: your statistics are still made available I see on https://kx1t.com/ais/stat.json whereas the HTML page tries to fetch the data from /stat.json which results in a fetch from https://kx1t.com/stat.json.

I changed the javascript fetch to "stat.json" and uploaded that as a patch. Perhaps this solves it? 

Thanks and regards, Jasper

kx1t, 2022-12-04:
I’ll rebuild and let you know — thanks for the very quick return and hup Holland hup :)

kx1t, 2022-12-04:
Works well now -- see https://kx1t.com/ais
Looking forward to seeing the rest implemented - ship lists, etc.
Adding "settings" there would be a bit dangerous if the website is exposed to the world - I may have to add Authentication on the reverse web proxy if there are no further restrictions implemented with AIS-catcher

I'll close the issue - thanks again for the ultra-fast fix!

closed 2022-12-04 by kx1t