jvde-github/AIS-catcher #46: web server syntax error

kx1t, 2022-12-10:
Hi Jasper,

This config line appears to give a Syntax Error:
```
/usr/local/bin/AIS-catcher -N 80 STATION "HELLO THIS IS A TEST" STATION_LINK https://www.shipxplorer.com/stations/SXTRPI000xxx -u 127.0.0.1 34994 -u 127.0.0.1 34995 -u 5.9.207.224 10xxx -u ais.vesselfinder.com 5xxx -u 144.76.54.111 35xxx -u 10.244.198.230 33333 -d AIS -gr tuner 40 rtlagc ON -H http://aprs.fi/jsonais/post/xxxxxx ID KX1T PROTOCOL aprs INTERVAL 30 RESPONSE off

```
Specifically:
```
Error: syntax error on command line with setting "-N". requires on or two parameters
```

Maybe there is another way I can tell it to use port 80 and also define STATION AND STATION_LINK?

(separately, and BTW -- we have an active Discord channel where we are discussing and admiring your AIS-Catcher software. If you're on Discord, feel free to join - https://discord.gg/9vB46pja )

kx1t, 2022-12-10:
I should add - I am at your `develop` branch at this commit: https://github.com/jvde-github/AIS-catcher/commit/1f7b22bd827e740472d98363b88d4a2b48a82c2d

kx1t, 2022-12-10:
never mind -- I recompiled once more and now it works.
Sorry for the inconvenience :)

closed 2022-12-10 by kx1t

jvde-github, 2022-12-10:
Thanks Ramon. Good to hear.. I btw regularly look at your station site  to see new functionality in action.  Helps to iron out the glitches so thanks for sharing that link. I will join the discord as well.