jvde-github/AIS-catcher #49: Dynamic PPM source enhancement

unxs0, 2022-12-31:
Provide a way to for AIS-catcher to get a new __-p_ value_ from some IPC method.

 Why? Since inexpensive RTL dongles (that developing nation users can better afford e.g. RTL2838) have a very wide crystal drift depending a lot on temperature among other factors. And since a new PPM value can easily be calculated by different means including using cell tower and even approximated by environment temperature. It would be useful to be able to set the _-p value_ that a running AIS-catcher instance is using. As long as it can be updated every 5 minutes without impacting performance it would be a great help.

Thank you for your consideration.
Cheers!

Taxom, 2023-01-23:
I do not think that this is necessary, the oscillator frequency drift is very small. Although it may be strongly shifted from the declared frequency. I have a cheap dongle on my ASDS-B feeder, which is more than 10 years old, the frequency shift on it is more than 50 ppm, but the temperature drift is not more than 1 ppm with a difference of temperatures from +5 to +60°C.

Taxom, 2023-01-25:
"There seems to be sometimes some dongles with serious drift: https://github.com/jvde-github/AIS-catcher-for-Android/issues/6"

I think this is just a defective quartz oscillator. Although if the addition of automatic frequency correction is not a problem, then it will not be superfluous. 

Thank you so much for your work!

unxs0, 2023-01-30:
Agree with all above. I can provide some supporting evidence of correlation between drift and case temp of dongle:

I collected 400 data points, case air temp vs ppm, of actual AIS messages using a common hi drift dongle. Data is from a fixed base station with high shipping traffic and loaded it into a spreadsheet and ran a correlation function getting 0.65 (1 would be perfect correlation).

[Spreadsheet Summary](https://github.com/unxs0/unxsVZ/blob/master/images/Screenshot%20from%202023-01-30%2018-20-58.png)

As shown in the image linked above I calculated a simple linear regression to create a slope intercept equation that I am using with success to reload AIS-catcher with a new error ppm every 5 mins if the calculation returns a different error ppm than the current one.

I am sure that the coding guru's here can create a much better automatic ppm correction scheme for high traffic areas. But from what I understand this would not be very useful to a lone fishing boat boat in Malasia with few messages. In that case the optional auto ppm correction switch could generate a formula for correction (correction setting probably non linear) based on an initial "calibration" run done in their busy home port.

Cheers and great work! I am comparing to comercial COMAR unit and getting excellent results.