jvde-github/AIS-catcher #51: Suggestion: Better documentation in README on and -h for .JSON config file nomenclature

eldplanko, 2023-01-08:
I’ve been using the json formatted config file, but have struggled with the appropriate nomenclature for the key/value pairs for each setting. It would be useful to include these in the -h output for each option. In the config file example in readme, you give examples examples on the sub-level attributes for the the rtlsdr, udp and http option, but it’s not obvious what available json key/values are for the other -h options.

g7ruh, 2023-01-10:
Jasper,

Thanks, noted and when you have time to do it is fine.

Roger

eldplanko, 2023-01-10:
Thanks, closing

eldplanko, 2023-01-11:
Can you confirm the syntax to enable biastee for RTL-SDR:

`AIS-catcher (build Jan  4 2023) v0.42
(C) Copyright 2021-2023 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
K",   "sample_rate": "1536K",   "biastee": false,   "buffer_count": 2  },  "serv
                                        ^
                                        ^
Error: syntax error in config file: "biastee" is not an allowed "key"
`