jvde-github/AIS-catcher #52: Select SDR by device name

SoroushRouhani, 2023-01-09:
Hello,

First of all, thank you for AIS-Catcher, which is great.
I have two DVBT sticks (RTL SDR). When I run the "AIS-catcher -l" command, the output is as follows, and the only component that can be used to separate the two is the Index, which unfortunately changes after rebooting the Raspberry Pi. Is it possible to add other information to distinguish the devices reliably? Like a device name or something you recommend?

Found 2 device(s):
0: Realtek, RTL2838UHIDIR, SN: 00000001
1: Realtek, RTL2838UHIDIR, SN: 00000001

Thank you again for your work. 
Soroush

OZ5RIX, 2023-01-09:
Hi Soroush 

You can use [rtl-eeprom](https://manpages.ubuntu.com/manpages/trusty/man1/rtl_eeprom.1.html) to change the SN of the devices:

Eg. To assign a new serial number to device 0, use the command `rtl_eeprom -d 0 -s 00000162`

Then use the command `AIS-catcher -d 00000162`  to start AIS-catcher.
 
I apologize for my English

SoroushRouhani, 2023-01-09:
Thank you for the solution.