jvde-github/AIS-catcher #53: Automatic Gain Control

Taxom, 2023-01-11:
Hi Jasper! Is it possible to implement the Automatic Gain Control (ADAPTIVE_DYNAMIC_RANGE and ADAPTIVE_BURST) by type of what is implemented in DUMP1090-FA? 

Thank you so much for the work!

Taxom, 2023-01-11:
I'm talking about it https://github.com/flightaware/dump1090/blob/master/README.adaptive-gain.md

Taxom, 2023-01-11:
It works very well for taking ADSB and since the types of signals and the conditions of reception are similar, in theory it should also work well with AIS.  
[image]

Taxom, 2023-01-11:
I also noticed something strange, immediately after the Ais-Catcher restart, there is a reception of distant and weak signals, but then they stop. AGC is turned off. Here are my settings:
` -d 00000162
 -v 10
 -M DT
 -gr RTLAGC off TUNER 38
 -a 100K
 -s 1536k
 -p -1
 -o 4
 -u 127.0.0.1 10111
`

Taxom, 2023-01-11:
9min after restart ais-catcher

[image]

Taxom, 2023-01-11:
30min after restart ais-catcher 
These are not ships, but basic stations and navigation buoys, so that the signal level from them should be approximately constant. 
[image]

Taxom, 2023-01-11:
I noticed this in the process of setting gain. It’s just that every time I change the gain level, I received a reception of weak signals, but after a short time they disappeared. I again changed of gain level and the situation was repeated, then I noticed that just restarting the Ais-Catcher is the same thing. 
I will still experiment with this and let you know how stable it is.

ROSTOCHE, 2023-01-20:
The software works very well with an RSP1A.
The only problem encountered is with a passenger Ro/Ro and therefore equipped with an A transponder.
I only see it very rarely, whereas other boats equipped in the same way and located just alongside are perfectly seen.
I've tried different values ​​at the LNAstate level with no noticeable change.
At the level of the table of tables I have seen twelve boats...
Is there a number limit for displayed boats?
Thanks in advance for the responses.