jvde-github/AIS-catcher #60: Ships page many columns and lot of white space

g7ruh, 2023-01-22:
Jasper

Rather than add a comment on the last issue about adding Bearing, a separate issue as it is not really related, it just triggered the thought.

There are a growing number of columns and a lot are far too wide for the data contained in them.

Will it be possible to select column width (like a spreadsheet) and store the layout? I am not too worried about the heading being truncated a bit: for example only seeing "Dist" of the heading "Distance" would be fine.

On the same subject, you could use "Brg" for "Bearing" as the bearing will never be more than 3 digits long.

I am not sure auto sizing column width would work, I prefer to see a fixed layout so I know where to look. As an example the RSSI column is slightly longer than the data field size but I would tend to leave it showing the full heading. Likewise "Lat" and "Lon" are shorter than the data field size so I would size it somewhat wider than the heading.

Thanks
Roger

g7ruh, 2023-01-23:
> I have adjusted the default width of the columns so presentation should be a bit more compact.

Jasper,

Thanks, that helps until you can provide user control.

In the 'Valid' column, if you changed 'Pending' to 'No' then that would enable a very narrow column when you provide user control. It is after all a binary column but I think Yes or No works better than T / F (True / False) or 0 / 1. 

About creating user defined areas, I am not sure. I use the distance / bearing to find a vessel, often, when zoomed out so I have an overall display of my coverage area. This means that there are a large number of vessels in a small area on the screen and the distance / bearing shows me roughly where to zoom in to look for the vessel I want.

Of course you could consider clicking on the vessel name in the Ships list and it would zoom in on the map to that vessel, that would be neat!

Roger

Taxom, 2023-01-23:
Could you change the view of the page, instead of the drop -down menu, make just a few tabs from above.
[image]

g7ruh, 2023-01-23:
> Of course you could consider clicking on the vessel name in the Ships list and it would zoom in on the map to that vessel, that would be neat!

For some reason it all stopped working like this, even after page refresh and closing and reopening pages, closing the browser and restarting. I will see if it happens again. I am using Firefox browser.

I thought it did work, from before. 

Quite likely browser related as it failed on both my live system and my development system. Live has V042, dev has current commit.

Roger

g7ruh, 2023-01-23:
My PC did get a reboot yesterday so Firefox would have had a clean restart. Problem was before that.

No error messages in console for either V042 or V042BE. If I see it again, I will check console and report back.
Both work as you say above.

Roger

g7ruh, 2023-05-08:
Jasper,

Small problem,

When I go to the ships list web page and select a column and select the right column boundary, it accepts the column boundary resize and then changes the page to the map display. On navigating back to the ships list web page it displays the resized column correctly. On selecting the next column....  and resizing.... it changes to the map page again.

I would expect it to allow multiple column resize (to get the display just how I want it) and then allow me to select adifferent web page to display.

Nearly there!

Regards
Roger

g7ruh, 2023-05-08:
One more thing:

When a using a narrow column... e.g. Bearing being 3 digits wide, the header has the heading (truncated) and the sort direction arrow overlayed

[image]

The blue header is deep enough to allow allow the direction sort arrow to be on a line below.

Regards
Roger