jvde-github/AIS-catcher #62: Message type 27 and ships table info

otm64, 2023-01-28:
Jasper,

I run a second instance of AIS-catcher for message type 27. I noticed the ships tables does not list vessel name or callsign. Is this because you pull the data from the AIS message and it is not present in type 27 or a bug in how type 27 is handled by your code? It would be nice to see vessel names.

Mike

[image]

tjswe, 2023-01-28:
Shipname is only present in message type 5,19 and 24. There has to be some local correlation between MMSI and shipname sourced from other frames or database.

https://gpsd.gitlab.io/gpsd/AIVDM.html#_type_27_long_range_ais_broadcast_message

otm64, 2023-01-28:
Many thanks for the explanations. I was hoping there was a central DB to easily pull from, but can see that is not the case. This is not super important to me, so however you address it (or not) in future releases is fine with me. I will play around with the netcat suggestion to get an aggregated map. You can close this issue as you see fit.