jvde-github/AIS-catcher #68: CPU use

Taxom, 2023-01-31:
After last update from 0.42-365 to 0.43-2 the loading of the CPU increased. 

[image]

g7ruh, 2023-01-31:
Jasper,
I have used the latest commit and find:

Using 'top' for CPU usage:

My live system using RTL dongle  V043 = 20% CPU latest commit 9% CPU
My test system using Airspy Mini V043 = 42% CPU latest commit 31% CPU.

I agree there was a CPU increase in the V043 release. Do not have more definitive measurements than above.

Hope that helps,
Roger

Taxom, 2023-02-01:
Yes it worked! Many thanks.
v0.44-18
[image]

Max