jvde-github/AIS-catcher #77: Wish list for web interface

anttinousiainen, 2023-02-11:
The new web interface is super nice addition, thanks for that!

Few minor feature wishes to make it even better:

-  "show/hide all tracks" button. Would be very useful to see by quick glance who is moving and who is not.
- support for own station coordinates to be dynamic, ie. when on a boat. Also auto centering/off-centering based on this would be needed
- 10/20/30min velocity and turning vectors for moving targets
- some way to combine multiple receivers into one web interface

I will think of few more!

anttinousiainen, 2023-02-11:
And while we are at it, in addition to PPM and RSSI, would be nice to report the received per message AIS-slot number as well. This would be useful in testing the AIS transponders.

anttinousiainen, 2023-02-11:
> The current way to combine multiple stations is to set up one instance that acts as a UDP server that listens on a particular port (-x .....). The other receivers send their messages with -u to this instance.

Yep, got it working this way, now my rtl is feeding A+B channels and AIrspy C+D.

I assume position for map is taken from A+B channels if available, not from msg 27, yes?

> What needs to be done still is to allow some station identification and pass through of meta data like Rssi and ppm.

Yes this would be great.

anttinousiainen, 2023-02-12:
> No, all lat/Lon from any message is used. We display the last position coming from any message.

Maybe the lat/lon from msg27 should be lower priority, if there are msg 1/2/3 lat/lon available.

It behaves interestingly otherwise:

[image]

anttinousiainen, 2023-02-14:
> You are right on msg 27. Have not seen the issue you show but it makes logical sense to use this info only when there is not more accurate data.

Yeh, this is somewhat annoying indeed.

[image]
[image]

anttinousiainen, 2023-02-14:
> How did you envisage btw to get more insights on these AIS slots you mentioned?

Just out of general interest, and also as some older (class B) units seem to have very "interesting" slot allocation mechanism, and seem to be stuck in a same slot for way longer than they are supposed to.

Also would maybe be beneficial to have a -vvv option to show also broken frames and unsuccesfully recovered, partial messages.

jortdejong, 2023-03-04:
As 'Anttinousiainen' mentioned dynamic 'Own Station Coordinates' should be a great improvement.
(support for own station coordinates to be dynamic, ie. when on a boat. Could be sent from UDP (nmea sentence) or hook directly to gps on serial/usb. Or could be just follow own MMSI, if one has AIS. Also auto centering/off-centering based on this would be needed)

anttinousiainen, 2023-03-10:
> In the new development branch we have the rudimentary functionality working. Clone it via

Nice progress.

I dunno if its too much, but share_loc on could have an option for mmsi_loc on and then give own ship mmsi to autofollow - or another ship/boat you want to follow from shore?

anttinousiainen, 2023-03-14:
> Once it looks stable it will be included in the next release.

Great!

btw. Have you given a thought on the receiving of all four channels ABCD at one go? 156.775 MHz to 162.025 MHz requires around 5.5MHz, so it would be doable with many SDRs.

anttinousiainen, 2023-03-23:
-s 2500k, random minute:
[AIS engine v0.44 #0-0]                 received: 42 msgs, total: 13422 msgs, rate: 0.699505 msg/s

-s 10000k -c ABCD every minute:
[AIS engine v0.44 #0-0]                 received: 0 msgs, total: 0 msgs, rate: 0 msg/s
[AIS engine v0.44 #0-1]                 received: 0 msgs, total: 0 msgs, rate: 0 msg/s

anttinousiainen, 2023-03-24:
> Is it an idea to run at 10 MHz at AB and CD individually? Might give a hint about interference from nearby frequencies.
> Btw I have been running it for a while now quite successfully on a R2 at 10 MHz.

There is indeed something funny going on with 6MHz and 10MHz. Same decent Diamond antenna I have been using before, and indeed there is almost no reception on higher sample rates. I tried moving the antenna around, no help.