jvde-github/AIS-catcher #99: Problem with the option -D for writing into a database

ROSTOCHE, 2023-04-23:
Good morning,

I want to use the option -D.
The first party of instructions seems to work very well (making of database ais and tables).

Just a small information. For the creation of tables you must be in the directory /usr/share/aiscatcher/AIS-catcher otherwise the command psql ais <DBMS/create.sql don't work.

The command sudo apt update and the installation of libpq-dev were made.

Now for me :::: i run aiscatcher as service.
When i put the option -D dbname=ais in my aiscatcher.conf and restart the service i encounter a problem.
Below i put a copy of the message: 

pi@raspberrypi:/usr/share/aiscatcher $ sudo systemctl status aiscatcher
● aiscatcher.service - AIS-catcher
     Loaded: loaded (/lib/systemd/system/aiscatcher.service; enabled; vendor pr>
     Active: activating (auto-restart) (Result: exit-code) since Sun 2023-04-23>
    Process: 4977 ExecStart=/bin/bash /usr/share/aiscatcher/start-ais.sh (code=>
   Main PID: 4977 (code=exited, status=255/EXCEPTION)
        CPU: 53ms

I don't find why.........

Coud you help me?

Best congratulations

Jean-Luc Rostini

ROSTOCHE, 2023-04-23:
Thank you very much for the quick response.

I have been using aiscatcher for several months and it works great.

I put some informations from cat /var/log/syslog

":228382600,"type":1,"nmea":["!AIVDM,1,1,,A,13IkGR00010`3W>GwP1I4A@800Rg,0*38"]}
Apr 23 10:29:04 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"B","rxtime":"20230423082904","signalpower":-63.5937,"ppm":0,"mmsi":228382600,"type":1,"nmea":["!AIVDM,1,1,,B,13IkGR00010`3W>GwP1I4A@800Rg,0*3B"]}
Apr 23 10:29:07 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423082907","signalpower":-59.1398,"ppm":2.02546,"mmsi":227001310,"type":18,"nmea":["!AIVDM,1,1,,A,B3HO3oP008:0<QUwidSQ3wSQnDeb,0*45"]}
Apr 23 10:29:07 raspberrypi aiscatcher[5101]: [AIS engine v0.45 #0-0]                 received: 3 msgs, total: 947 msgs, rate: 0.299326 msg/s
Apr 23 10:29:08 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423082908","signalpower":-59.5182,"ppm":2.02546,"mmsi":227001310,"type":24,"nmea":["!AIVDM,1,1,,A,H3HO3oTtCBD4Gb46Gjjmh00`?44t,0*19"]}
Apr 23 10:29:10 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423082910","signalpower":-70.0872,"ppm":2.02546,"mmsi":227345990,"type":18,"nmea":["!AIVDM,1,1,,A,B3Hl6AP0B8:09FUw`P1qCwTUoP06,0*35"]}
Apr 23 10:29:13 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423082912","signalpower":-64.7782,"ppm":-0.868056,"mmsi":228382600,"type":1,"nmea":["!AIVDM,1,1,,A,13IkGR00000`3VnGwP1I4A@H0<0B,0*53"]}
Apr 23 10:29:13 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"B","rxtime":"20230423082912","signalpower":-34.8567,"ppm":1.15741,"mmsi":228382600,"type":1,"nmea":["!AIVDM,1,1,,B,13IkGR00000`3VnGwP1I4A@H0<0B,0*50"]}
Apr 23 10:29:13 raspberrypi aiscatcher[5101]: {"class":"AIS","device":"AIS-catcher","channel":"B","rxtime":"20230423082913","signalpower":-59.886,"ppm":-3.47222,"mmsi":227915930,"type":18,"nmea":["!AIVDM,1,1,,B,B3IFpVP0?89wwgUwT=ms;wVQl0SC,0*16"]}

Now it's the results when i launch start-ais.sh in a terminal window

AIS-catcher (build Apr 22 2023) v0.45e-50-g08cab1c
(C) Copyright 2021-2023 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

in your response i see that the user of database must be aiscat and  this user doesn't exist in Postgresql.....
Searching for device with SN 1807027893.
UDP: open socket for host: 5.9.207.224, port: 12096, filter: OFF, JSON: OFF
UDP: open socket for host: 192.168.1.169, port: 10110, filter: OFF, JSON: OFF
UDP: open socket for host: 127.0.0.1, port: 10111, filter: OFF, JSON: OFF
Server: reading statistics from stat.bin
HTML Server running at port 8383
Server: starting backup service every 5 minutes.
Device    : RSP1A
Settings  : rate 2304K format CF32 agc ON lnastate 5 grdb 40 
Model #0  : [AIS engine v0.45] ps_ema ON afc_wide OFF droop ON fp_ds OFF 
{"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423085157","signalpower":-50.4495,"ppm":0.578704,"mmsi":329019040,"type":18,"nmea":["!AIVDM,1,1,,A,B4qif`0008:0E7Uwq4wQ3wd5oP06,0*53"]}
{"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423085202","signalpower":-53.8011,"ppm":2.60417,"mmsi":227410840,"type":18,"nmea":["!AIVDM,1,1,,A,B3Hp3V000H:0@M5wp2>sowQ5oP06,0*7D"]}
{"class":"AIS","device":"AIS-catcher","channel":"A","rxtime":"20230423085203","signalpower":-36.5525,"ppm":1.73611,"mmsi":228382600,"type":1,"nmea":["!AIVDM,1,1,,A,13IkGR0v@10`3V0GwP694AH60<0B,0*3B"]}
{"class":"AIS","device":"AIS-catcher","channel":"B","rxtime":"20230423085203","s

there's no error

But the user aiscat is not created like user of Postgresql.

I try this and see what happens.

Best regards

Jean-Luc

ROSTOCHE, 2023-04-23:
I don't understand what happen.

I've created the user

postgres=# \du
                                              Liste des rôles
 Nom du rôle |                                    Attributs                                    | Membre de 
-------------+---------------------------------------------------------------------------------+-----------
 aiscat      | Superutilisateur, Créer un rôle, Créer une base                                 | {}
 postgres    | Superutilisateur, Créer un rôle, Créer une base, Réplication, Contournement RLS | {}

and the database

postgres=# \l
                                  Liste des bases de données
    Nom    | Propriétaire | Encodage | Collationnement | Type caract. |    Droits d'accès     
-----------+--------------+----------+-----------------+--------------+-----------------------
 ais       | aiscat       | UTF8     | fr_FR.UTF-8     | fr_FR.UTF-8  | 
 postgres  | postgres     | UTF8     | fr_FR.UTF-8     | fr_FR.UTF-8  | 
 template0 | postgres     | UTF8     | fr_FR.UTF-8     | fr_FR.UTF-8  | =c/postgres          +
           |              |          |                 |              | postgres=CTc/postgres
 template1 | postgres     | UTF8     | fr_FR.UTF-8     | fr_FR.UTF-8  | =c/postgres          +
           |              |          |                 |              | postgres=CTc/postgres
(4 lignes)

Probably i miss something.

best regards

J-L

ROSTOCHE, 2023-04-25:
Good morning,
By guiding myself with the indicated help I managed to do what I wanted.
At first I was a little scared by what the aiscatcher service refused to restart... It came that I simply had not been to the end.
Now I have all the information in the tables.
All I have to do is interconnect with my QGIS program to retrieve the data in shapefiles.
Thank you very much for the quick answers.
Best regards.

Jean-Luc Rostini

jnunezortuno, 2023-06-06:
Hi all! 

I have a working installation of AIS-catcher in Linux. I have a problem with the Postgre DB connection feature. When I try:

_sudo -u aiscat AIS-catcher -D dbname=ais -v_

I become following response:

_AIS-catcher (build Jun  5 2023) v0.47-4-geab9490
(C) Copyright 2021-2023 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Found Rafael Micro R820T tuner
Error: DBMS: no support for PostgeSQL build in._

Any idea of what's going on?

jnunezortuno, 2023-06-06:
I have used the Linux script from abcd567a (https://github.com/abcd567a/install-aiscatcher/blob/master/README.md) but it doesn't includes libpq-dev.

Have also figthed with the Postgre DB user permissions but now is finally working!

Thanks!

abcd567a, 2023-06-06:
I have now added following in my install script at https://github.com/abcd567a/install-aiscatcher

`sudo apt install -y libpq-dev`

caliKev, 2023-06-07:
Is there a way to get the database working in a docker install? I’m a bit of a novice with Postgres db. Can I modify the dicker file to launch posgres db?

abcd567a, 2023-06-09:
> I have now added following command in my install script at https://github.com/abcd567a/install-aiscatcher.
> 
> It will install `libpq-dev` _**before**_ running cmake and building AIS-catcher
> 
> `sudo apt install -y libpq-dev`