jvde-github/AIS-catcher #106: Temporary drop in Messages

au51gtm, 2023-05-07:
Hi,
I have an issue where the number of messages will drop to roughly half for a period of approximately thirty minutes and then recovers again.  This seems to happen once every 24-36 hours, sometimes less frequent.  I've tried two different RTL-SDRs and multiple builds on the RPi, but it seems persistent.  AIS-catcher runs on a dedicated RPi and is not used or disturbed for any other purpose.  My apologies if this question has been addressed elsewhere, but my quick search didn't show anything.  Screenshots are attached.
[image]
[image]

au51gtm, 2023-05-08:
> Hello, thanks for raising.
>
> Given the testing you have done, I think it might be some interfering
> broadcast of a nearby signal that raises the noise level to ~dB (as
> signals with a level below that are not picked up during this timeframe).
>
> You could try to inspect the signal with other tools like SDR# during
> and after the signal drop period and see if there is any broadcast nearby.
>
> What settings are you using? You could try to experiment with |-a
> 192K| (bandwidth filter), (auto) gain settings or some hardware
> filter. But as you experience this only for 30 minutes every 24/36
> hours, might not be worth that investment.
>
> Thanks,
> Jasper

Hello Jasper,

Thanks for your very quick response and for the tips for further testing 
and tracking down this issue.

I will try and catch it the next time it has the drop in message count 
and monitor for adjacent interference signals.

I'm using a fixed TUNER gain (42.1) with RTLAGC off and the default 
bandwidth.

First, I will look for interfering signals and if that doesn't show any 
culprit, then I will try the auto gain and bandwidth settings that you 
have suggested.

Kind regards,

Trevor.

otm64, 2023-05-09:
I have a similar issue that occurs every four hours. In my case it is caused by a weather/traffic report broadcast at 161.9 Mhz off a 300 meter tower 7 km from my receiver. The strong signal knocks back collection when it broadcasts. Nothing much for me to do but just accept it and wait for it to pass - usually 5-15 minutes.

au51gtm, 2023-05-09:
Thank you Jasper and OTM64.

The cause of the drop in messages certainly seems to be massive 
interference.  It seems to be very broadband in nature, and it will take 
me a while to identify the source.  It also seems to have quite random 
timing and so I don't know if I will ever actually identify the source, 
but I am going to try.

I have SDR++ running with an RTL-SDR tuned to 161.975 and I do see an 
occasional transmission around 161.6 and 161.7 but they don't cause any 
issues.  In the image attached, I was able to capture the end of the 
interference, which stops instantly and the noise floor drops from 
approximately -75, back to the normal level of -90.

I've now set up a second SDR/SDR++ instance and will try and use it to 
locate the interfering signal.

In the mean time, I will just accept that I am in a noisy environment 
for AIS.  I have other SDRs capturing ADS-B, RadioSonde and High 
Altitude Balloon signals and fortunately, they don't appear to be effected.

otm64, 2023-05-12:
If the periodicity appears random, then Jasper's comment could be true. I can verify that from my own experience. My antenna and receiver are indoors and thus I've learned first hand about all the indoor QRM which can affect VHF reception. Unfortunately the list of culprits is almost endless, one needs to keep watch on the signal levels and if it drops think "what just turned on (by itself or my turning a switch)?" and investigate further. A short list of my main issues ...

- non-dimmable LED bulb controlled by a dimmer switch kills my signal. Replacing with a legit dimmable LED helped a lot.
- lamps which have a on/off switch imbedded in the cord. Adding ferrite beads to the power cords helped.
- certain appliances are bad news for me - vacuum cleaner and washing machine (especially when its motor runs at variable speed to agitate the wash)

Also depending on your living arrangement, it may not be you but a nearby neighbour which is causing the issue. Fortunately all my QRM is self-inflicted and not from outside.

In the end I treat the hunt for the elusive QRM to be a sport and an ancillary learning experience. Radio and interference go hand in hand and we have to deal with it as it arises. Best of luck in your search.

-

au51gtm, 2023-05-13:
Thank you OTM64, for the extensive information and some really great 
tips, based on your own experience.

This information will come in very handy as I carry out my search.  My 
AIS receiver is located in the garage, away from the house by a distance 
of about 8m, and the antenna is mounted on top of a 9m pole.  The 
neighboring houses are very close by and so it's quite realistic to 
expect that the interference could be coming from any of them.

At the moment, the biggest challenge is actually catching it in the act.

Hopefully, I will be nearby when the next event occurs and will be able 
to run some tests that advance the investigation.

au51gtm, 2023-06-25:
Hello Jasper and otm64,
Well after a long and frustrating search for the source of the interference, I finally identified the culprit.
It is being caused by LED lights, as was suggested by otm64.  After spending weeks logging when the interference appeared, I could see an alignment with our morning and evening activities.  After eliminating any movement detection devices, cameras etc., I started looking closely at other things we turned on at those times.  I just happened to be checking AIS Catcher one morning when my wife switched on the bathroom light and the messages were instantly halved.  I removed all the LED lights then refitted a normal filament lamp and the problem disappeared.  Re-installing any one LED lamp caused the problem to reappear.  They are dimmable LEDs used in a non dimming situation so they should be okay, I guess.  Anyway, this week I will get some replacements and hopefully they will not cause the interference.  We have similar LED bulbs of other manufacturers throughout the house and they do not seem to cause any issues at all.  Thank you both for your advice and guidance.

otm64, 2023-06-26:
Glad to hear you were able to find the culprit. Yes, in addition to the dimmable vs. non-dimmable situation, the undefinable definition of a "quality" LED is an issue. Sometimes it is worth the extra money for a name-brand bulb, in other cases a cheap no-name bulb from the supermarket works just fine. You never really know until you screw it in, which is the frustrating bit.