jvde-github/AIS-catcher #108: Oddity with Ships table and shipnames.pjs

michaelgeorge001, 2023-05-17:
Version v0.45e-224-ga216a65 appears to have issues with [shipnames.pjs](https://github.com/jvde-github/AIS-Catcher-PLUGINS/blob/main/shipnames.pjs).

The Ships table will not display custom names or callsigns, whereas the Maps has no issue with the custom data.

I do know that AIS-catcher version v0.45e-26-g20f8a66 does display the info correctly.  Unfortunately I missed a lot of revisions in between so I can't identify when the issue started.

However, is it possible that when Tabulator was integrated into AIS-catcher, that it may be the source?

Thanks,
Mike

michaelgeorge001, 2023-05-17:
Hi Jasper, all is good now. The fix worked - the custom names and callsigns display correctly in the Ships table.

Thanks,
Mike

Taxom, 2023-05-30:
It seems to me that the problem is still relevant, I still do not receive the names of some vessels, which previously displayed normally.
It would be nice if you could see not only the last message, but several, for example, for the last hour or the last ten messages.
Thanks!

Taxom, 2023-05-31:
This ship early displayed as MU23 but now it's empty "ship name" have only callsign.

[image]
[image]
[image]
[data(1).csv](https://github.com/jvde-github/AIS-catcher/files/11608173/data.1.csv)
[ships_array.json.txt](https://github.com/jvde-github/AIS-catcher/files/11608182/ships_array.json.txt)

Taxom, 2023-05-31:
This ferry also early displayed normal, but now it's not.
[image]

 -d 00000162
 -v 10
 -M DT
 -gr RTLAGC off TUNER 26
 -go AFC_WIDE on
 -a 100K
 -s 1536k
 -p -2
 -o 5 filter on block_type 1,2,3,4,5,9,18,19,21,24
 -u 127.0.0.1 10111
 -u 192.168.1.145 10103 JSON on
 -u 192.168.1.100 10101
 -u 5.9.207.224 xxxxx
 -u 195.201.71.220 xxxx
 -u 144.76.54.111 xxxx
 -u 31.172.90.245 xxxx
 -u 144.76.105.244 xxxx
 -u 178.162.215.175 xxxx
 -u 148.251.96.197 xxxxx
 -N 8383 FILE stat.bin BACKUP 60 ABOUT about.md MESSAGE on
 -N STATION MaxPi_AIS_Toronto STATION_LINK https://www.myshiptracking.com/stations/station?id=736
 -N PLUGIN_DIR /usr/share/aiscatcher/plugins
 -N LAT 43.62 LON -79.48 share_loc on
 -N REUSE_PORT on

Loaded plugins:
JS: /usr/share/aiscatcher/plugins/add_rings.pjs
JS: /usr/share/aiscatcher/plugins/google_option.pjs
JS: /usr/share/aiscatcher/plugins/my_maps_with_no_labels.pjs
JS: /usr/share/aiscatcher/plugins/vesselfinder_option.pjs
JS: /usr/share/aiscatcher/plugins/marinetraffic_option.pjs
JS: /usr/share/aiscatcher/plugins/myshiptracking_option.pjs
JS: /usr/share/aiscatcher/plugins/google_option.pjs
JS: /usr/share/aiscatcher/plugins/latlon_in_dms.pjs
JS: /usr/share/aiscatcher/plugins/shipxplorer_option.pjs
JS: /usr/share/aiscatcher/plugins/vesselfinder_option.pjs

Taxom, 2023-05-31:
And something can interfere with receiving a message of type 5, since in some cases there are more than 2000 reports but without type 5? It seems to me earlier I got the name of these ships much faster.  And now I have half of the ships even after several days of continuous working of aiscatcher only MMSI are displayed.

Taxom, 2023-06-06:
I tried to disable the AFC_wide function and now I get the names of the vessels.
But unfortunately now I do not see vessels who has a strong shift in the frequency of the transmitter (more +-5ppm).

Taxom, 2023-06-07:
[image]

Taxom, 2023-06-07:
These vessels are not visible when the AFC_wide function is turned off.
[image]

Taxom, 2023-06-07:
It looks like it works. Most vessels were determined quickly enough.
Thanks!

: Settings  : rate 1536K bw 100K freqoffset -2.000000 format CU8 tuner 26.000000 rtlagc OFF biastee OFF buffer_count 2
: Model #0  : [AIS engine v0.47] ps_ema ON afc_wide ON droop ON fp_ds OFF

[image]

Taxom, 2023-06-07:
Yep, Previously(when i used AFC_WIDE), I needed to use this file in order to see the names of most ships:
"
//=============
//My ship names base
// /home/plugins/usr/share/aiscatcher/plugins/shipnames.pjs

const shipNames = { 
	//mmsi:		{name:"name"},
	//mmsi:		{name:"name",callsign:"callsign"},

	316045069:	{name:"SAM MCBRIDE"},
	316005305:	{name:"KAJAMA"},
	316002873:	{name:"OCEAN GOLF"},
	316009316:	{name:"M.R. KANE"},
	316048671:	{name:"HIAWATHA"},
	316023266:	{name:"MU03"},
	316023267:	{name:"MU04"},
	316023269:	{name:"MU11"},
	316035651:	{name:"MU20"},
	316023273:	{name:"MU21"},	
	316023274:	{name:"MU22"},
	316023275:	{name:"MU23"},
	316050404:	{name:"DAVID HORNELL"},
	316032594:	{name:"IRON GUPPY"},
	316225000:	{name:"EMPIRE SANDY"},
	316048477:	{name:"ODYSSEY TORONTO"},
	316045066:	{name:"ONGIARA"},
	316048758:	{name:"YANKEE LADY4"},
	848507543:	{name:"ENDURANCE"},
	316045081:	{name:"WM INGLIS"},
	316045082:	{name:"THOMAS RENNIE"},
	316048672:	{name:"KWASIND"},
	316048588:	{name:"EMPESS OF CANADA"},
	316045085:	{name:"NORTHERN SPIRIT1"},
	316044324:	{name:"ORIOLE"},
	316049426:	{name:"YANKEE LADY 3"},
	316030796:	{name:"WM. THORNTON"},
	316046964:	{name:"ESCAPE T.O"}
	};

function getShipName(ship) { return (ship.mmsi in shipNames && shipNames[ship.mmsi].hasOwnProperty("name")) ? shipNames[ship.mmsi].name : ship.shipname; }
function getCallSign(ship) { return (ship.mmsi in shipNames && shipNames[ship.mmsi].hasOwnProperty("callsign")) ? shipNames[ship.mmsi].callsign : ship.callsign; }
"