jvde-github/AIS-catcher #133: postgreSQL

bucolo, 2023-07-17:
Hi,
Thank you very much for aiscatcher !
Can i use a external database ? it is possible ?

i try  "-D dbname=ais host=192.168.1.x user=aiscatcher password=xxxxx STATION_ID 78" with no success

Thanks
Brice
(https://www.marinetraffic.com/en/ais/details/stations/10668)

mdkendall, 2023-07-17:
The docs say you need quotes around the connection string if it contains spaces. Try this:

`AIS-catcher -D "dbname=ais host=192.168.1.x user=aiscatcher password=xxxxx" STATION_ID 78`

bucolo, 2023-07-17:
I had already tried with quotes, it didn't work. but with postgresql:// it works fine. you must url encode the password

thank you very much

bucolo, 2023-07-30:
Hello again
I have been using an external Postgres database for a few weeks now, it works very well except when there is a network outage or the database server restarts. the connection to the database is definitely lost but aiscatcher continues to operate.
Is it possible to restart the connection regularly in case of loss of connection?
Thanks !

bucolo, 2023-08-01:
Thank you very much it works, but I think it took a while to restart the connection after the network cut

Aug  1 01:31:34 piadsb aiscatcher[12589]: DBMS: write completed (10218 bytes).
Aug  1 01:31:44 piadsb aiscatcher[12589]: DBMS: write completed (11498 bytes).
Aug  1 01:31:54 piadsb aiscatcher[12589]: DBMS: write completed (14087 bytes).
Aug  1 01:32:04 piadsb aiscatcher[12589]: DBMS: write completed (11756 bytes).
Aug  1 01:32:14 piadsb aiscatcher[12589]: DBMS: write completed (13036 bytes).
Aug  1 01:32:24 piadsb aiscatcher[12589]: DBMS: write completed (11017 bytes).
Aug  1 01:44:03 piadsb aiscatcher[12589]: DBMS: writing to database slow or failed, data lost.
Aug  1 01:48:21 piadsb aiscatcher[12589]: DBMS: Error writing PostgreSQL: could not receive data from server: Connection timed out
Aug  1 01:48:23 piadsb aiscatcher[12589]: DBMS: Error writing PostgreSQL: server closed the connection unexpectedly
Aug  1 01:48:25 piadsb aiscatcher[12589]: DBMS: Connection to PostgreSQL lost. Attempting to reset...
Aug  1 01:48:25 piadsb aiscatcher[12589]: DBMS: Connection successfully reset.
Aug  1 01:48:25 piadsb aiscatcher[12589]: DBMS: write completed (1382 bytes).
Aug  1 01:48:35 piadsb aiscatcher[12589]: DBMS: write completed (14418 bytes).

I'll do another test and try to give more details.

bucolo, 2023-08-01:
> 
> I will also add a new table that contains one record per vessel with the latest data.

that's what I need. My request would be simpler :)

bucolo, 2023-08-01:
I confirm, it works but it's long, 10 minutes at least to reestablish the connection

Aug  1 13:22:35 piadsb aiscatcher[12589]: DBMS: write completed (2138 bytes).
Aug  1 13:22:37 piadsb aiscatcher[12589]: DBMS: write completed (5044 bytes).
Aug  1 13:22:39 piadsb aiscatcher[12589]: DBMS: write completed (2941 bytes).
Aug  1 13:22:41 piadsb aiscatcher[12589]: DBMS: write completed (3311 bytes).

            13:22 network KO (disable rule in firewall)
            13:28 network OK (enable rule in firewall)

Aug  1 13:32:10 piadsb aiscatcher[12589]: DBMS: writing to database slow or failed, data lost.
Aug  1 13:38:27 piadsb aiscatcher[12589]: DBMS: Error writing PostgreSQL: could not receive data from server: Connection timed out
Aug  1 13:38:27 piadsb aiscatcher[12589]: DBMS: Error writing PostgreSQL: server closed the connection unexpectedly
Aug  1 13:38:29 piadsb aiscatcher[12589]: DBMS: Connection to PostgreSQL lost. Attempting to reset...
Aug  1 13:38:29 piadsb aiscatcher[12589]: DBMS: Connection successfully reset.
Aug  1 13:38:29 piadsb aiscatcher[12589]: DBMS: write completed (2381 bytes).
Aug  1 13:38:40 piadsb aiscatcher[12589]: DBMS: write completed (15497 bytes).
Aug  1 13:38:50 piadsb aiscatcher[12589]: DBMS: write completed (16972 bytes).
Aug  1 13:39:00 piadsb aiscatcher[12589]: DBMS: write completed (15346 bytes).

bucolo, 2023-08-01:
Thank you, it's ok for the new table 👍 
[image]