jvde-github/AIS-catcher #162: I still have some problems with the stability of feeding some AIS aggregators.

Taxom, 2023-09-05:
I still can’t locate this problem. Both aggregators get a UDP stream from me.
[image]
utc time
[image]
local internet traffic from Pi
[image]

i don't see a problem with the streaming on other aggregators, but they don't have such detailed statistics. 
[image]

Taxom, 2023-09-05:
I changed to TCP both. I will continue monitoring it.

otm64, 2023-09-06:
Same here for MarineTraffic. I am guessing this has to do with the recent acquisition of FleetMon & MT by Kpler. I got an e-mail from FleetMon last week saying their development team was merging with MarineTraffic's and asking me to approve them sharing my data with MT (I already feed MT). I got the vibe the trend is to migrate everything to a MT-centric product, so I cancelled my FleetMon account & stopped feeding them. No use sending data to the same ultimate organization twice. What we are experiencing may be issues processing data on the back end as they merge two systems together.

Taxom, 2023-09-07:
Yes, it looks better, but now there is much less traffic than it was last weekend when this problem appear.
(45-50 ships now and more than 100 then)

But there are still passes at (also tcp) https://www.myshiptracking.com/stations/station?id=736
[image]
But perhaps this problem is on their servers.

Taxom, 2023-09-09:
Yesterday when i used the TCP connect from two mobiles (OpenCPN and AIS-catcher-android) and (OpenCPN), there was a lag again on the Marinetraffic statistic.  I could not make a screenshot of graph, but it was a short time like two spikes.

g7ruh, 2023-09-10:
@Taxom 
I noticed these small gaps when I had router / ISP issues. It was when Jasper and I were tracking the UDP outage issues.

If I disconnected my data phone connection to the router and plugged it back in, nothing changed router-wise and the router re-synced in a few seconds. This gave things similar to your graph above on the MT site.

Because I had so many disconnects triggered by a noisy line, it helped Jasper to provide the Persist setting for a TCP connect, and I had a good testbed for it. 

Just an idea,
Roger

obama, 2023-09-18:
I just tried sending data to MT and it says he got my message but the ships i found dont show up on the map. But a ship i have near me says it was updated few minutes ago, but not by me (had id turned off) so the map refreshes... ill try -P.

Taxom, 2023-09-19:
I think that this problem can be closed as solved. I no longer see systematic problems with the feeding of AIS aggregators. Those small bursts that are still periodically happening are most likely associated with Internet providers or on aggregators' servers. If I find something, I will write here or open a new topic.

closed 2023-09-19 by Taxom