jvde-github/AIS-catcher #170: OpenSSL errors when running Cmake

otm64, 2023-09-16:
I attempted to update to the latest build today, but am receiving the following errors when running cmake. Both openssl and libssl3 packages are installed on my system and the latest version.

**-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR) 
-- OPENSSL: not found - OPENSSL_INCLUDE_DIR-NOTFOUND,** 
-- Checking for module 'libcurl'
--   Package 'libcurl', required by 'virtual:world', not found
-- CURL: not found - CURL_INCLUDE_DIR-NOTFOUND, CURL_LIBRARY-NOTFOUND
-- ZLIB: found - /usr/include, /usr/lib/x86_64-linux-gnu/libz.so
-- Checking for module 'libpq'
--   Package 'libpq', required by 'virtual:world', not found
-- PQ: not found - PQ_INCLUDE_DIR-NOTFOUND, PQ_LIBRARY-NOTFOUND
-- Checking for module 'soxr'
--   Found soxr, version 0.1.3
-- SOXR: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsoxr.so
-- Checking for module 'samplerate'
--   Found samplerate, version 0.2.2
-- libsamplerate: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsamplerate.so
-- RTLSDR: found - /usr/local/include, /usr/local/lib/librtlsdr.so
-- RTLSDR: bias-tee support included.
-- RTLSDR: tuner badwidth support included.
-- SDRPLAY: not found.
-- Checking for module 'libairspy'
--   Package 'libairspy', required by 'virtual:world', not found
-- AIRSPY: not found.
-- AIRSPYHF: found - /usr/local/include/libairspyhf, /usr/local/lib/libairspyhf.so
-- Checking for module 'libhackrf'
--   Package 'libhackrf', required by 'virtual:world', not found
-- HACKRF: not found.
-- Checking for module 'libzmq'
--   Package 'libzmq', required by 'virtual:world', not found
-- ZMQ: not found.
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
OPENSSL_CRYPTO_LIBRARY (ADVANCED)
    linked by target "AIS-catcher" in directory /home/mike/AIS-catcher
OPENSSL_SSL_LIBRARY (ADVANCED)
    linked by target "AIS-catcher" in directory /home/mike/AIS-catcher

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR) 
-- OPENSSL: not found - OPENSSL_INCLUDE_DIR-NOTFOUND, 
-- Checking for module 'libcurl'
--   Package 'libcurl', required by 'virtual:world', not found
-- CURL: not found - CURL_INCLUDE_DIR-NOTFOUND, CURL_LIBRARY-NOTFOUND
-- ZLIB: found - /usr/include, /usr/lib/x86_64-linux-gnu/libz.so
-- Checking for module 'libpq'
--   Package 'libpq', required by 'virtual:world', not found
-- PQ: not found - PQ_INCLUDE_DIR-NOTFOUND, PQ_LIBRARY-NOTFOUND
-- Checking for module 'soxr'
--   Found soxr, version 0.1.3
-- SOXR: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsoxr.so
-- Checking for module 'samplerate'
--   Found samplerate, version 0.2.2
-- libsamplerate: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsamplerate.so
-- RTLSDR: found - /usr/local/include, /usr/local/lib/librtlsdr.so
-- RTLSDR: bias-tee support included.
-- RTLSDR: tuner badwidth support included.
-- SDRPLAY: not found.
-- Checking for module 'libairspy'
--   Package 'libairspy', required by 'virtual:world', not found
-- AIRSPY: not found.
-- AIRSPYHF: found - /usr/local/include/libairspyhf, /usr/local/lib/libairspyhf.so
-- Checking for module 'libhackrf'
--   Package 'libhackrf', required by 'virtual:world', not found
-- HACKRF: not found.
-- Checking for module 'libzmq'
--   Package 'libzmq', required by 'virtual:world', not found
-- ZMQ: not found.
-- Configuring done
**CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
OPENSSL_CRYPTO_LIBRARY (ADVANCED)
    linked by target "AIS-catcher" in directory /home/mike/AIS-catcher
OPENSSL_SSL_LIBRARY (ADVANCED)
    linked by target "AIS-catcher" in directory /home/mike/AIS-catcher

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.**

Taxom, 2023-09-16:
https://stackoverflow.com/questions/15476123/missing-cmake-variable-when-compiling-with-openssl

otm64, 2023-09-16:
Yes, I have no need for SSL just wanted the build to execute without stopping. I recompiled and all works as usual now. Thanks for the quick fix!