jvde-github/AIS-catcher #171: Maps with labels gone

anoo2niem, 2023-09-17:
Hi,

I just updated and it seems openstreetview with labels is gone.
It was the first option before update.
Now i have 'Voyager', 'Sattelite'; 'Positron', 'Dark matter'.

Is there any way i can bring this back, with another plugin ie. maps with labels?
Can't seem to find this anywhere.

Another unrelated question, but does there exist a plugin to use metric dimension and/or speed?

Thanks for this wonderful piece of software!

anoo2niem, 2023-09-17:
Nevermind, just found https://github.com/jvde-github/AIS-Catcher-PLUGINS/blob/main/moremaps.pjs.
Placed it in my-plugins folder and it brought back openstreetmap with labels.
Removed dimension_in_feet.pjs aswell and it reverted dimensions back in metric.

If there is, which i doubt, any plugin to show speed in km/h, please let me know.
Thank u!

If not, this issue can be closed.

anoo2niem, 2023-09-17:
Wonderful, bedankt Jasper!

Another unrelated question, at the moment my sdr is connected to my proxmox server, so my antenna aswell is mounted inside. Which gives me a range of aprrox +- 30km.

I'm thinking of mounting my antenna & sdr outside, with a raspberry pi using usbip over wifi in a weatherproof birdhouse.
Would this be recommended or would i have some kind of loss aswell using this solution?

Alteratively i could install AIS-catcher directly on the raspberry pi ofcourse, but i would prefer it in a lightweight vm on my proxmox server, if above solution would be viable.

Thanks again!

Taxom, 2023-09-17:
This will of course work, but what's the fundamental meaning? AIS Catcher works great on Raspberry Pi if you want, you can run one copy on the Raspberry without a web interface that will feed the second copy on your server on the UDP or TCP protocol.

anoo2niem, 2023-09-17:
@Taxom @jvde-github 

At the moment i have 1 sdr & 1 hackrf. I ordered 2 more sdr's. At the moment i have 3 lightweight vm's configured with AIS-catcher, tar1090, and PagerMon (pocsag messages) all working one at the time with the one sdr. (SDR v4, i found i have better reception with v4 compared to hackrf). For pocsag messages hackrf seems to work as good as sdr, but to be able to feed it into pagermon i have to use gqrx udp stream it to netcat, then pipe it in sox... which results in +-80% higher cpu usage on the vm than sdr. So i'll go with 3 sdr's.

The initial plan was to mount a raspberry pi in a birdhouse with the 3 sdr's and use usbip on the proxmox host to the respective vm's. I thought that would rule out potential package conflicts etc. compared to if i would install all 3 on the pi. The additional benefit of scheduled backup's of the vm's and to be able to deploy them anywere would also be nice. Also i thought it would reduce the rate at which the sd-card at a given moment would fail.

I will look into your proposition to run the pi as a AIS-catcher, tar1090 & pagermon server, and run the clients in the vm, but i think this method also has the potential to increase conflicts, compared to just mounting the usb ports  on the pi to the respective vm's with usbip.

I guess i will indeed have to experiment a little bit and see what works best.

Thanks again Jasper for the rapid implementation to use metric.

Taxom, 2023-09-17:
I have Orange Pi  Zero 2 with two RTL-SDR dongles one for AIS catcher and another for dump1090-fa. I'm feeding many ADS-B (9) and AIS (8) aggregators and 2-3 local users.
This is my CPU graf 
[image]
Two spikes are Ais-Catcher compilation and updates. As you can see, the processor load is about 25% at 800 messages per second ADS-B and 10-20 messages per second AIS.

This is my setup.
[image]

http://taxom.ddns.net:8383

anoo2niem, 2023-09-18:
@Taxom 

Very nice setup!
I might go that route aswel.
Gave me some idea's to implement in the birdhouse.

Thanks for the insight & sharing!