jvde-github/AIS-catcher #181: Send NMEA to serial port

yellow-sky, 2023-10-06:
In some cases, it may be necessary to send messages not over the network, but through a serial port. Now, as far as I understand, this cannot be done.

yellow-sky, 2023-10-18:
So, when using docker, this approach has some difficulties, but is possible. I will try. But imho this is a useful feature

closed 2023-10-18 by yellow-sky

yanniedog, 2024-02-01:
This is much needed. Please fix.

If you don't have a serial output device to test on, perhaps something like approach could be a workaround?
https://stackoverflow.com/questions/2291772/virtual-serial-device-in-python?noredirect=1&lq=1

yanniedog, 2024-02-01:
My apologies, perhaps I misunderstand the OP. I was referring to receiving NMEA via the serial port.