jvde-github/AIS-catcher #224: SDRPlay Buffer Overrun

g1oob, 2024-01-18:
Hello, New User to AIS-Catcher. After watching the Tech Minds video. Wanted to try this program out.
Successfully compiled and installed the latest version of AIS-Catcher on a Mac, running Monterey OS. 
Using an SDRPlay RSPdx SDR unit, the API 3.07.03 correctly complied with the Github install commands. 
Checked to see if the SDR uit can be seen with the command AIS-catcher -l, all good the unit was present. 
Then ran the next command to start the AIS decoding, AIS-catcher -v 10 -u 127.0.0.1 10110 -u 127.0.0.1 10111. 
The result :- 
AIS-catcher (build Jan 18 2024) v0.56-21-g2edf4415
(C) Copyright 2021-2023 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Warning: sample rate 2304K upsampled to 3072K.
UDP: open socket for host: 127.0.0.1, port: 10110, filter: OFF, JSON: OFF
UDP: open socket for host: 127.0.0.1, port: 10111, filter: OFF, JSON: OFF
Device    : RSPDX
Settings  : rate 2304K format CF32 agc OFF lnastate 0 grdb 32
Model #0 -> (Src: 1, Grp: 1): [AIS engine v0.55] ps_ema ON afc_wide ON droop ON fp_ds OFF dsk OFF
!AIVDM,1,1,,A,13aPOO3P01wjVQBM>J2t1wvJ2H7j,0*0D ( MSG: 1, REPEAT: 0, MMSI: 244850556)
!AIVDM,1,1,,B,15SGeb000?wiRsNMDBlb8`:J2H7j,0*06 ( MSG: 1, REPEAT: 0, MMSI: 372633000)
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.
SDRPLAY: buffer overrun.

Endless SDRPLAY: buffer overrun. Repeating.

Is there an option to stop the 'Buffer Overrun' from happening ? 

The webserver works as far as displaying a webpage, but tried to view anything thinking the Buffer Overrun
would be neglected does not work, AIS-catcher -v 10 -u 127.0.0.1 10110 -M DT -N 8100. With port 8100. 

So this 'Buffer Overrun' stops anything from going forward. 

Is there a work around using an SDRPlay unit ?

Thanks.

mpuz, 2024-01-23:
i had this when tried to run aiscatcher on raspberry 1 with 512mb of ram, looks like it cant address memory

g1oob, 2024-01-24:
Hello jvde, Unfortunatley the issue for the endless 'Buffer Overun' messages maybe that the machine I tried to install it on is a non-native Apple variant. Its an HP Elitedesk G3 800 with MacOS Monterey, using Clover to boot the EFI and Opencore to 'Fake' the drivers that the OS uses. This 'Fake' driver for USB works ok on other SDR apps like CubicSDR and SDRPlay Connect. How knows what the issue is, but it could be OpenCore related ? I switched machine to another HP G3 800 and succesfully installed it on Linux Mint 21, no 'Buffer Overrun' messages.