jvde-github/AIS-catcher #231: User friendly setup script

yanniedog, 2024-02-03:
Would it be possible to provide a setup script which helps novice users configure this correctly? Ie, so that it takes users through a basic configuration, step by step, explaining each point. Also, some values could be automatically determined/detected by the script and offered as a "suggested value" based on how they have their system setup.

A significant proportion of people using your software have very limited experience of Linux, java, command line parameters etc.

This would dramatically lower the barrier to entry for your software, and allow many more people to use it. This should not require a computer science degree to set up. Your parents and grandparents should be able to set this up on their own, without any additional help.

Just a suggestion.

This is a philosophy using with my AI project, called Autocrew. If you're interested, check out my repo.

Cheers!

Taxom, 2024-02-03:
https://github.com/abcd567a/install-aiscatcher

yanniedog, 2024-02-03:
Exactly my point.

yanniedog, 2024-02-03:
Unfortunately, the app isn't much good to me because I'm using a SLR350Ni. 
Automated, script-based installation is the key. Users should just be able to click "GO" and it automatically get's it working without further user input. Keep it nice and simple for stupid people like me :)