jvde-github/AIS-catcher #253: select a group of vessels to track

g7ruh, 2024-04-08:
Jasper,

As you know, I live in an area renowned for yacht racing.

It is interesting to watch the tactics develop. 

Would it be possible to enable a group of vessels to have tracking enabled? E.G. use a mouse button to create a rectangle and set tracking on for all vessels within that rectangle.

select an area as below

[image]

and enable tracking for all vessels:

[image]

Of course the idea came when I had to select all one at a time!

The track length needs to be variable, it is too short for this use case:

[image]

and from OpenCPN which allows track length to be configured

[image]

Thanks for considering this,
regards
Roger

g7ruh, 2024-04-08:
Of course, the arrow indicating speed and direction on the OpenCPN map adds more to the experience, as a real-time status

Roger

g7ruh, 2024-04-08:
[image]

the highlight when hovering over the vessel icon is very good at showing the individual tactics.

Roger

g7ruh, 2024-04-08:
Jasper,

the race in the example went on for most of the afternoon, so the tracks were very long. if a bit thick on the default OpenCPN track width, it made a 'mess' of confusion on the screen, however your feature to highlight when hovering on a vessel picks out the detail nicely.

To that end, I think that the length of track (time-wise) needs to be a configurable setting, for normal use as it is or a bit longer is OK, for tracking a race and tactics, it needs to be a lot longer, but configurable.

Maybe AIS-catcher needs to have the option to record tracks for the "selected" rectangle or all tracks for e.g. 24 hours so the tracks are recoverable, post event, at least for a short while. E.G. playing back this afternoon's Cowes Week races: there are many separate races happening in The Solent at the same time. That option may win you some users in the yacht racing world too. Replay analysis tool, the tactician's friend!!!!!!!

Roger

kit-powell, 2024-04-11:
This is right up my alley, as I want to do something similar (not for racing per se, but for fishing)

I’m not sure how fast I can get it done but if y’all would like I’d work on a companion platform to create a sort of geo-temporal fence and replay the tracks (probably leaflet based) with varying speed and chart/map/imagery)

Though I freely admit my github experience is limited to one project I put up years ago

gharbarka, 2025-04-21:
hi,

as a yacht racing enthusiast I am interested in an application using ais-catcher for tracking and recording a list of boats based on names or MMSI number for playback later offline for the purpose of analysing strategies of individual boats. This or something similar is intimated in the above discussion. Has any progress been achieved since the last message was posted a year ago?  Or perhaps a related project may have been created. Thanks for any insight.

gharbarka, 2025-04-21:
Thank you very much for your speedy reply. First of all allow me to congratulate you for a very nifty piece of software.

Yes I have read issue: #341 but it seems to add little to the problem. I have also read the wiki section related to the use of PostgreSQL which is interesting but to me somewhat of an overkill for the purpose i have in mind. Also, as I understand it you still have to process that database further to get to an end result. The issue of needing to be connected to the internet to use the database would also be an issue for on the water analysis.

I wonder if it is possible to send the  !AIVDM output to a decoder which can extract the positions and perhaps the speed of selected boats  with a time stamp to a simple .csv file for further processing.

gharbarka, 2025-04-22:
Indeed I had already started going down that route. I am currently learning how to process json files for futher processing in Python. The json concept is new to me.

What would be more useful would be the possibility of processing the json format in real time and then only save fields of interested related to a preposted list of MMSI numbers.