jvde-github/AIS-catcher #256: AIS-SART Alert

NR857, 2024-04-16:
Send an alert when an AIS SART is activated within a receiving station's coverage area

Phil353556, 2024-04-16:
Hi,
Interesting
https://en.wikipedia.org/wiki/AIS-SART
"Once per minute, the position is sent as a series of eight identical position report messages (four on 161.975 MHz and four on 162.025 MHz)."

NR857, 2024-04-16:
AIS SART's have MMSI's commencing 970, so perhaps that string could be used to identify activations. I'm new to programming and Github etc, so I don't know how to this yet lol

NR857, 2024-04-16:
https://cdn.standards.iteh.ai/samples/16694/7bf752a5752548be8abe3a3b90f71445/IEC-61097-14-2010.pdf

NR857, 2024-04-18:
Online AIS SART NMEA message decoder. May be useful 

https://rl.se/aivdm

Phil353556, 2024-04-18:
on the aiscatcher map there is already something isn't it?
You should ask @jvde-github about that.
on the screenshot, the green rectangle,  when the mouse is over, the text displayed is "SARTE/EPIRB" 
- SART beacon 
- An EPIRB fixed distress beacon is a transmitter that emits a signal in the 406 MHz frequency band.
Kind regards
[image]

NR857, 2024-04-18:
I didn't notice that.  Which means that the Catcher already identifies the AIS SART class. What's needed to send a push notification or popup when one is activated? Under the regulations SARTs are required to be tested regularly I have seen alerts for tests on Boat Beacon, but obviously you will see more alerts in heavy traffic areas such as Gibraltar Straits which is not too far from here.

Agreed email alert on Web would be great.

All very interesting!

Phil353556, 2024-04-19:
@jvde-github 
@NR857 I found a document you may be interested by: https://gpsd.gitlab.io/gpsd/AIVDM.html
I think this is a complete document on AIS message.

If I loop back to your first question @NR857 "Send an alert when an AIS SART is activated within a receiving station's coverage area" 
Do you thing that a mail should be sent?

NR857, 2024-04-19:
Yes I think a mail might bve easiest?

Phil353556, 2024-04-19:
@jvde-github 
What about the same thing ie send an email,  in case of MOB?

NR857, 2024-04-19:
Ok if the MOB or crew activate an AIS SART, a different matter if it's 406 or 121.5

Phil353556, 2024-04-20:
Great,  thank you.  will see what happens.

NR857, 2024-04-20:
Noted: ELLI GLITSCHER 211413680 reported "Aground" 15:50Z Hamburg

https://aiscatcher.org/alerts

Nice work

Phil353556, 2024-04-20:
@jvde-github On the same topic ie SAR:  what means exactly "SAR aircraft" on the webpage accessible localy from the station webserver?
Thank you
Philippe
[image]

Phil353556, 2024-04-20:
Ok, thank you,  I didn't know that SAR messages from aircrafts was received by AIS's stations

NR857, 2024-04-20:
Re SAR aircraft 

Some use "111" ahead of the country code, followed by an identifier.

i.e. SAR 111232532 airborne 4 minutes ago over Scotland or SAR 111257007, airborne now over Råde, Norway.

Spanish SAR helicopters sometimes use callsign "HELIMER" and also the 111 designator,  followed by country code 224, then three digit identifier.  You can find these on any of the public sources Marinetraffic,  Vesselfinder, AISHUB etc 

Irish SAR aircraft use SAR, tgen 250, such as SAR 250002898, currently airborne off south east Ireland.

NR857, 2024-04-20:
A peculiarity of Irish and UK coastguard organisations is that they regularly carry out mountain rescue,  snow relief and inland air ambulance missions- watch out for them!

Phil353556, 2024-04-27:
I understand, however  a SAR is SAR whatever the size of the vessel. isn't it?

g7ruh, 2024-04-28:
Could also consider identifying SAR heli squawk code: "0023 Aircraft engaged in actual SAR Operations" from the ADSB message if you have access to ADSB decode data, e.g. take the data feed into AIS-catcher and monitor for specific squawk codes. Or write a AIS-catcher utility to parse the data and only feed the selected codes to AIS-catcher. That would save burdening AIS-catcher with mostly "useless" data with respect to ship tracking.

Apart from non-sea missions, also training as per here, lowering man onto deck training. Apologies Jasper, this was from before I had AIS-catcher! 

[image]

NR857, 2024-05-17:
> For that the ais2adsb project might be interesting in combination with
> Virtual Radar Server. Also worked recently with the author of tar1090 to
> get ships in the ADSB viewer.
>
> I will close this item. Base logic is implement. We can always open a new
> issue if there are some new ideas.
>
> Thanks all.

👍