jvde-github/AIS-catcher #259: Binary Message 8

MatsA, 2024-04-26:
Hi and once again thanks for a very nice application.

Nowadays I'm also using the app in our boat together with Signal K and the web app Freeboard. Using this new combination weather reporting ATON:s appeared. Se pictures. According to my knowledge these are not displayed in your web app and maybe could be a nice add-on ? 

Googled a little and found, below, in a product specification the format for message 8.
http://www.ohmex.com/AtoNpaper.pdf

/Mats
[image]

[image]

MatsA, 2024-04-27:
God Morning, up and working early! 😉

Nice you found examples. Think it can be nice additional info in the web app.

/Mats

MatsA, 2024-04-27:
The pictures above is from the web app Freeboard.
Source found here https://github.com/SignalK/freeboard-sk which maybe can help

/Mats

MatsA, 2024-04-27:
Nice ! 

Yes I think your proposal is nice.

> "The question now is where and how to integrate it without too much clutter and work. Perhaps we can add an icon to the map where we receive binary messages, user clicks on it and shows the above in a table....."

If you want to minimize the clutter maybe a filter so no AtoN:s is shown ?

MatsA, 2024-04-27:
OK

Regarding the "stream".... maybe you should add information at the https://aiscatcher.org/stations that you must be on version >=  v0.58 to participate..... I did the mistake to be on a lower level and therefore there are 2 stations with name "Domsten Harbour". Have a nice weekend, Mats

MatsA, 2024-04-27:
Absolutely, feel free to remove it.

Phil353556, 2024-07-24:
Hi, @flygfantast 
My 2cts
To decode  I use this website : https://www.maritec.co.za/aisvdmvdodecoding
[image]
kind regards