jvde-github/AIS-catcher #267: local dashboard: messages per hour and day

g7ruh, 2024-05-11:
Jasper, I notice that the vessel count on the messages per hour and per day "flat line" as they are too low relative to the message count.

I guess they need a scale on the RHS of the graph like you did on aiscatcher.org.
As you can see, another busy (sunny and light winds) day in The Solent:

[image]

Now 700 vessels!

[image]

Regards
Roger

g7ruh, 2024-05-11:
Let me know when and I will test.

Also now you have vessels per minute in messages per minute, I guess we don't need the separate vessels per minute graph?

g7ruh, 2024-05-11:
Thanks Jasper,

I will test and see what I think. The other graph takes up space and is no longer needed. For a while I have thought that and having a day or week related view on number of vessels is more informative than just per minute.

Let me know when you tweak it next and I can have a look.

Roger

g7ruh, 2024-05-11:
Hi Jasper,
thanks (as always) for the high speed response! Having installed the latest version....

I agree, orange is wrong, it needs to be a high contrast change from the colours of the base graph as it is different data, not directly related to that graph. It also gets lost in the other graph colours. 

I would say red, but bearing in mind that a large number of the population are red-green colour blind, maybe black is good. 

The black line is the other scale and colours are the message count: easy to remember and for the brain to interpret as very different, therefore the other scale. 

I had a salutary experience many years ago developing prototype software for a large organisation where we showed finance changes in green for up and red for down.... the senior person in that part of the finance business said " yes very good but I am red-green colour blind so they both look " grey" to me.

I agree the scales are a bit "large" so (pun) scaling them down is better. I agree 60K or 600K is much better: it makes the graph area bigger: the whole point of a graph!

hope that helps
Roger

Taxom, 2024-05-11:
> You can view on your online dashboard: ![image](https://private-user-images.githubusercontent.com/52420030/329812068-d0d3ea4f-ed2f-4417-9ae3-196ca6cc4a36.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0NjQwMzIsIm5iZiI6MTcxNTQ2MzczMiwicGF0aCI6Ii81MjQyMDAzMC8zMjk4MTIwNjgtZDBkM2VhNGYtZWQyZi00NDE3LTlhZTMtMTk2Y2E2Y2M0YTM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTExVDIxNDIxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyZmRhODNmMmNkMTY2YjA2YmFkZDRlY2ZhOGExNTM1ZjU5ZGM1MzQxMzk4N2VkNWQyNjI2ODM4OGE0NmRhY2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GulNN8-6U6JxFgjOUZGSuUOZ9QZ4od8CyIQPF6FO2qc)
> 
> I think the orange is a bit difficult to see (so perhaps will make a darker color) and don't like the space taken by the large numbers (so perhaps say 600K instead of 600,000)....

Maybe use a science format like "1e5"

g7ruh, 2024-05-15:
> Maybe use a science format like "1e5"
@Taxom 

Trouble is that I suspect your average butcher, baker and candlestick maker and sailor too may not understand the format but the "K" shorthand is known by lots of folks now because of exposure to things computing.

Roger

Taxom, 2024-05-16:
> > Maybe use a science format like "1e5"
> > @Taxom
> 
> Trouble is that I suspect your average butcher, baker and candlestick maker and sailor too may not understand the format but the "K" shorthand is known by lots of folks now because of exposure to things computing.
> 
> Roger

Okay, well then lowercase would be more correct. 1k not 1K