jvde-github/AIS-catcher #268: aiscatcher.org dashboard info

g7ruh, 2024-05-13:
Jasper,

Please can you change the clock to 24 hour format. It is confusing to see the below 12 digits after noon. 24 hour is more universal format.

[image]

In this example I though that it had not updated since early this morning, it took a bit of thinking to < LF + CR > in my head to find the PM indicator!

Thanks, Roger

g7ruh, 2024-05-13:
Thank you, well my head says thank you, as a radio amateur !!!

Roger

Phil353556, 2024-05-13:
Hi,
I agree, 24h format is better 
Thank you @jvde-github 
[image]