jvde-github/AIS-catcher #269: aiscatcher.org coverage map

g7ruh, 2024-05-13:
Jasper,

the coverage map defaults to a zoomed in setting, quite close in to my station:

[image]

what I have to do, each time I look at the coverage, is to zoom out two or three steps to see the overall area:

[image]

in the above an extra zoom "click", so a bit wider than I normally do.

Can this be changed to show a wider area by default: "the coverage area". If this is done it would also show wider reception caused by tropo enhancement (for example), as an automatic view on local or enhanced propagation. 

Apologies if  you have this in plan as I know you are still working on this functionality.

Roger

Phil353556, 2024-05-14:
Hi,
Fully agree with @g7ruh 
default view:
[image]

zoom out twice:
[image]

Phil

g7ruh, 2024-05-14:
Looks like Jasper may have just worked his magic again Philippe:

[image]

Thanks Jasper

Roger