jvde-github/AIS-catcher #335: Open Shipcard

pingel20, 2024-11-30:
Clicking a vessel always opens a minimized shipcard. (v0.6)

Is there a way to have the shipcard always opened full size?

Thanks.

g7ruh, 2024-11-30:
@pingel20 

You can select each line on the drop down and the box changes colour (slightly) to indicate selection. 

Call up the shipcard and select the down arrow to bring up all items

Click all lines and then the full set opens each time you select the shipcard. Also useful to select several items in case you want more than one, but not all items.

Hope that helps,

Roger

pingel20, 2024-12-02:
Currently, when I click a ship icon the shipcard opens with (only) the vessel name. But I see this name already on the map. So no real extra info after clicking. Only after clicking the ship card header I see all info.

For me (and others?) it would be nice when the shipcard opens with all info displayed.

pingel20, 2024-12-03:
I have it working now. Selecting all rows for one ship will also show all rows when clicking a next ship icon.
That's how I like it. :)
Thanks.