jvde-github/AIS-catcher #371: Edge version info icon two data values

g7ruh, 2025-03-11:
Jasper,

When I hover over the white circle or white "i" I get 

[image]

When I hover over the blue inner I get 

[image]

I recompiled to the current Edge version prior to this test. Same before and after re-compile.

Initially on program start up, hover shows a single value on hover over both areas.

Current weather in the local area  wind NE about 13 kt gust 16kt. The blue one has wind at 124 deg (wrong, I think) and the white has 30 deg, about right. Unfortunately the Nab Tower has no online data available (assume radio link down) but the tower is stll sending AIS data via RF.

The screen grabs are a minute or so apart.

After another screen reload the wind is 1 kt different and air temp is 0.1 deg different, wind however is 23 and 29 kt. This aligns with the OpenCPN reported data 14 kt 29deg (NNE) , air 8 deg, pressure  1004.

South Bramble only has water temp data at present (any chance of adding water current speed and direction too?) it shows different readings for hover over blue and white areas: 8.6 blue and 8.8 white.  On reload both blue and white are 8.8. OpenCPN also reports 8.8 deg.

Is the data on one of them slightly older than the other area, rather than displaying the same values, perhaps?

Roger

g7ruh, 2025-03-11:
Thanks, Jasper,
Happy to test, as always. Let me know when.

I noticed it because I was checking one of them and saw a change in digits and wondered why! Then I discovered that the blue bits were different from the white bits: 'does not compute' was the result.

Roger

g7ruh, 2025-03-18:
Jasper, thanks for the update, let me know when you would like some testing.

Distractions AIS-catcher-wise OK, When "Real-Life" gets in the way... well that takes priority over everything, including AIS-catcher!

Regards
Roger

andlil, 2025-03-22:
Also, slightly related, when using it on mobile, there is no way to hover so the ability to click would be appreciated.