jvde-github/AIS-catcher #376: Home Assistant - Addon

CzarofAK, 2025-03-27:
I like your project!

May i drop a feature request? It would be really cool to have this as an Home Assistant Addon available!

CzarofAK, 2025-03-28:
@jvde-github, i have no clue about SW, so bare with me :-)
MQTT is of course no problem in HA, but this only pulls metadata in and does not provide a nice GUI/Map. On the other hand, i still would require a second device to run your containers.

My usecase is to run everything on my Home Assistant Yellow in my Motorhome. So i would like to stick with one computer device.

I tried to run Portainer on my HA OS and installed there your dockers which worked. But as Portainer is overly powerfull and not official supported, that will also not be my solution.

Last but not least, there is a possibility to run you dockers as a Addon on HA, see here: https://developers.home-assistant.io/docs/add-ons/
As it uses docker too, i think it would not be such a big task to fiddle it together (but as i said, i am not a SW guy!!).
- The advantage for a Addon is, it is basically a one click installation for users like me

CzarofAK, 2025-03-28:
This is an example which i use in my installation. From what it is doing, it is very closely related to your SW.
https://github.com/MaxWinterstein/homeassistant-addons/tree/main/adsb-multi-portal-feeder

CzarofAK, 2025-03-29:
Yes, that would be very easy. but it really does not like running Portainer to add the ais-catcher.

CzarofAK, 2025-04-12:
I tried around with AI to create this Addon, and it kind of worked...
https://github.com/CzarofAK/homesassistant-addons/tree/main/ais_catcher_control
As I have nothing to do with coding, it might be good that a professional could have a quick look.

Four things still to be investigated:
1. The RTL-SDR was working fine, but I had issues with my Cypho-150
2. What i like to add is, that the addon is pulling the GPS data from the HomeAssistant like this addon: https://github.com/MaxWinterstein/homeassistant-addons/tree/main/adsb-multi-portal-feeder
3. Also an configuration field for the webserver port, as this is pretty usefull
4. Would be nice to have also the webinterface to be shown in the sidebar

May you can have a quick look, @jvde-github

jvde-github, 2025-12-12:
Thanks for sharing. 

FWIW. In principle we can add additional JSON schemas (where lat is called latitude etc) by adding a column to https://github.com/jvde-github/AIS-catcher/blob/main/Source/JSON/Keys.cpp. Could be interesting if there is sufficient demand.

Elwell, 2025-12-12:
I don't think extra entries are required in the json as Xrlls method of using the template is quite elegant.

Unrelated tho, you dont need to publish the cr/nl in mqtt. I'll raise a bug/patch for that