jvde-github/AIS-catcher #385: cmake 4.x: Configuration fails with `Compatibility with CMake < 3.5 has been removed from CMake.`. Workaround: `-DCMAKE_POLICY_VERSION_MINIMUM=3.5`.

dreirund, 2025-04-23:
Ahoj,

with cmake 4.0.1, some changes did occur that break the `cmake` run for this package.

I get the following error when doing `cmake -B build -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr .`:  
```
CMake Error at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
```

Adding `-DCMAKE_POLICY_VERSION_MINIMUM=3.5` does work around this, but as I got from some other project, the `cmake` configuration code/ options should be updated to cater for `cmake` version 4.x.

I am building release version 0.61 of this software.

Regards!

dreirund, 2025-04-23:
> Ok, have added something to the CMake file. The usual build process should work again, I hope. But if not, let me know. Thanks for flagging.

I can test when there is a new release and the [AUR package](https://aur.archlinux.org/packages/ais-catcher) has been updated to the new release.

dreirund, 2025-04-23:
I can confirm that [this patch](https://github.com/jvde-github/AIS-catcher/commit/5a36a2f6a1a08b6760b41174d6de75660f532126.patch) makes it working.

Regards!

closed 2025-04-23 by dreirund

jvde-github, 2025-10-21:
Works for me, can you perhaps show the output, as need bit more to reproduce:
```
:~/AIS-catcher/build$ ~/cmake-4.1.2-linux-x86_64/bin/cmake ..
-- Using CMake 4.1.2 with 4.0+ features
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Looking for syslog.h
-- Looking for syslog.h - found
-- Looking for syslog
-- Looking for syslog - found
-- Looking for linux/can.h
-- Looking for linux/can.h - found
-- Looking for linux/can/raw.h
-- Looking for linux/can/raw.h - found
-- NMEA2000: not found  NMEA2000_LIB-NOTFOUND NMEA2000_INCLUDE-NOTFOUND
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Checking for module 'libssl'
--   Found libssl, version 1.1.1f
-- OPENSSL: found - /usr/include, /usr/lib/x86_64-linux-gnu/libssl.so, /usr/lib/x86_64-linux-gnu/libcrypto.so
-- Checking for module 'zlib'
--   Found zlib, version 1.2.11
-- ZLIB: found - /usr/include, /usr/lib/x86_64-linux-gnu/libz.so
-- Checking for module 'libpq'
--   Found libpq, version 12.22
-- PQ: found - /usr/include/postgresql, /usr/lib/x86_64-linux-gnu/libpq.so
-- Checking for module 'sqlite3'
--   Found sqlite3, version 3.31.1
-- SQLite: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Checking for module 'soxr'
--   Found soxr, version 0.1.3
-- SOXR: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsoxr.so
-- Checking for module 'samplerate'
--   Found samplerate, version 0.1.9
-- libsamplerate: found - /usr/include, /usr/lib/x86_64-linux-gnu/libsamplerate.so
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 2.0.1
-- RTLSDR: found - /usr/local/include, /usr/local/lib/librtlsdr.so
-- Looking for rtlsdr_set_bias_tee
-- Looking for rtlsdr_set_bias_tee - found
-- RTLSDR: bias-tee support included.
-- Looking for rtlsdr_set_tuner_bandwidth
-- Looking for rtlsdr_set_tuner_bandwidth - found
-- RTLSDR: tuner badwidth support included.
-- SDRPLAY: not found.
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- AIRSPY: found /usr/include/libairspy, /usr/lib/x86_64-linux-gnu/libairspy.so
-- Checking for module 'libairspyhf'
--   Found libairspyhf, version 1.4
-- AIRSPYHF: found - /usr/include/libairspyhf, /usr/lib/x86_64-linux-gnu/libairspyhf.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.5
-- HACKRF: found - /usr/include/libhackrf, /usr/lib/x86_64-linux-gnu/libhackrf.so
-- Checking for module 'libzmq'
--   Found libzmq, version 4.3.2
-- ZMQ: found - /usr/include, /usr/lib/x86_64-linux-gnu/libzmq.so
-- Configuring done (4.9s)
-- Generating done (0.0s)
-- Build files have been written to: /home/jvdeshof/AIS-catcher/build
````

dreirund, 2025-10-23:
Yes, my error.

Details: [Here](https://aur.archlinux.org/packages/ais-catcher#comment-1044632).

Regards!